VUnit/vunit #13: Topical Videos

barri, 2015-01-26:
I would like to see more presentations on YouTube. More specifically videos that are around five minutes long which cover specific use cases of VUnit. This would be very useful when you are trying to integrate VUnit into your current workflow or just not too familiar with the tool.

Keep up the good work!
Barri

LarsAsplund, 2015-01-26:
This work has started. First will be a short introduction to unit testing and then a short introduction to VUnit. Please provide suggestions on topics that you would like to see and subscribe to the YouTube channel to get posted when new videos are added.

barri, 2015-02-17:
One use case that I would find interesting to see presented is the following. The DUT or a driver connected to the DUT is fed with data from a file. The resulting data output from the DUT is then compared to another file containing the expected/desired output. I will try to think of some other simple but common use cases.

Thanks!
Barri

kraigher, 2015-02-17:
@barri In my project we do this routinely. Should probably be an example in the example/ folder rather than a video though. Maybe I will extend the array-example to do this.

LarsAsplund, 2015-02-17:
Agree that it should be an example but one thing doesn't exclude the other... I'm just about to add two new videos tonight based on my presentation at FPGA Forum in Norway. The first is about why unit testing is useful, the other a short intro to VUnit. After that I would like to add a few different ways to approach VUnit, e.g.:
- Just using the automated compile support for any type of project
- Adding VUnit test runner when you already have self-checking testbenches (using asserts, Bitvis Utility Library, OSVVM or some other solution out there)

Once I have the overview videos and the getting started videos I will move to the topic videos. Your input will go into that work..

kraigher, 2015-02-18:
@barri I updated examples/array to also feature reading an input file and comparing the result with an output file.

kraigher, 2015-04-01:
@LarsAsplund @barri close this issue?

LarsAsplund, 2015-04-01:
@kraigher Yes, I think so. Better to have specific issues for each requested video. An issue like this will never be closed

closed 2015-04-08 by kraigher