VUnit/vunit #14: Redistribute OSVVM with VUnit

LarsAsplund, 2015-01-28:
The OSVVM functionality for random value generation and functional coverage is commonly used with VUnit and it has been suggested that it should be redistributed with VUnit for a more easy access of the latest version. OSVVM is often shipped with the simulators but to get the latest version you have to check the osvvm.org site for updates and signup to download. OSVVM already has a redistribution repo on Github but it isn't maintained.

The OSVVM and VUnit licenses allow such redistribution and it has also been okeyed by the OSVVM maintainer Jim Lewis.

The latest release of OSVVM, 2014_07a, can be redistributed "as is". All you need is to add an osvvm library and to add all VHDL source files in the OSVVM directory to your Python script.

There is also an upcoming version 2015_01 which is currently in a beta stage. This version will add logging and checking functionality to OSVVM. The result of their checking (called alerts in OSVVM) mechanisms can be injected into the VUnit framework if the OSVVM checking mechanism is preferred over what is provided with VUnit. Some minor details regarding this integration is discussed with the maintainer. When 2015_01 becomes publicly available it will be added to VUnit along with an example/acceptance test showing how the integration is performed.

LarsAsplund, 2015-02-03:
OSVVM 2014_07a is now part of the VUnit distribution. OSVVM 2015_01 will be added once it becomes official.

kraigher, 2015-02-03:
I would be nice to add a ui.py:VUnit method such as 'add_osvvm' to conveniently add osvvm to the compile list.

LarsAsplund, 2015-02-03:
Yes, that's probably a good idea. It's not very many lines today, for example
```python
osvvm = ui.add_library('osvvm')
osvvm.add_source_files(join(dirname(__file__), '..', '..', 'vhdl', 'osvvm', '*.vhd'), [])
```
The tricky thing to remember is the empty list at the end of the add_source_files call. Whithout that globally enabled preprocessors will be applied to OSVVM as well. Since OSVVM cointains its own log calls the location preprocessor will trigger which leads to compilation errors.

kraigher, 2015-02-04:
I added the method now.

LarsAsplund, 2015-02-04:
Good move to make the library name configurable. I prefer the osvvm default but I've seen both osvvm and synthworks being used in code.

LarsAsplund, 2015-02-06:
OSVVM 2015_1 was just released and I've updated our repo. I'm going to add an example on how their alerts (checks) can be used together with VUnit and then close this issue.

LarsAsplund, 2015-02-09:
Integration examples in examples/osvvm_integration.

closed 2015-02-09 by LarsAsplund