VUnit/vunit #15: Infinite loop in check_equal when converting > 32 bits to a decimal integer.

kraigher, 2015-02-02:
The following code causes an infinite loop when converting the 'expected' value to decimal integer due to undefined bits:
```vhdl
check_equal(std_logic_vector'(x"10"), std_logic_vector'(x"1000000-"));
```

By the way: Would we want the 'expected' value to contain 'dont care' bits meaning they should not be checked? Kind of like the std_match vhdl function defined in numeric_std: http://en.wikipedia.org/wiki/Numeric_std

LarsAsplund, 2015-02-02:
I fixed the bug. A metalogical value will now result in NaN (not a number) when printed as an integer. The suggestion to handle don't cares as in std_match seems like a reasonable enhancement. Let me think about this a bit more.

LarsAsplund, 2015-02-03:
Using std_match in check_equal instead of the "=" operator causes some problems, e.g.:
- x"ZZ" = x"ZZ" is true but std_match(x"ZZ", x"ZZ") is false which is not what you expect if you associate checking for equality with the "=" operator as the name suggests
- check_equal(9, "100-") type of calls make the implementation harder but is probably not the main use case
- The main use case is that you have a well-defined bit field which you compare to an expected value where some bits are don't care, e.g. "1001" matches "1--1". However, if you expect a well-defined bit field and get a value with don't cares you normally wouldn't consider that a match, i.e. "1--1" doesn't match "1001". This means that the use case isn't symmetric but std_match is.

The best solution is probably to keep check_equal as is and then add new checkers:
- check_match is a wrapper around std_match and is symmetric. What you get over check(std_match(x,y)) is the convenience of automatic error message generation. 
- check_expect allows don't cares but is not symmetric as explained above

These two checkers can only compare vector types, not integer types

LarsAsplund, 2015-02-14:
Changed my mind on check_expect. Unless there is a strong request for this check I think we should exclude it. The difference to check_match is rather subtle and I haven't find a good name to express that difference. Also, the use case with don't cares in the tested value is not very common and can be handled in other ways, e.g. `check_match(to_01x("1--1"), "1001")` would fail.

commit 4e526933 (2015-02-14): Added check_match. Closes #15.
closed 2015-02-14 by LarsAsplund