VUnit/vunit #16: The flag -modelsimini is necessary on Modelsim Microsemi v10.3a

tomasnilefrost, 2015-02-06:
The fix for Issue #12 (5917f349810b968dc8aac307eabf8429a40c1d01) breaks vunit for "Model Technology ModelSim Microsemi vsim 10.3a Simulator 2014.04 Apr 15 2014"

After the fix it is impossible to load a design since vsim cannot find the compiled libaries (cannot find the modelsim.ini-file)

By reintroducing the flag -modelsimini in vunit_load() the majority of the acceptance tests can be run successfully.
However, this breaks issue #12 since -modelsimini cannot be specified correctly with spaces in the path.

kraigher, 2015-02-06:
As you mentioned there seems to be no way to get Modelsim 10.3 to accept spaces in the argument to the -modelsimini flag of the vsim command. Unlike the -wlf flag which would accept an escaped path containing spaces. Thus I removed the -modelsimini flag beliveing that the MODELSIM environment variable would work for everyone. So you are saying you version does not respect the MODELSIM environent variable? Is there some way to direct your version of modelsim to use a modelsimini with spaces in the path? I would like to have a solution which will work for everyone.

tomasnilefrost, 2015-02-06:
At first I thought that the version didn't respect the MODELSIM environment variable but I now rather think that it has something to do with how the variable is set.

If I rename the output "end to end out" to "end_to_end_out" and prefix the unittest-run with the correct modelsimini-path it still doesn't work.
$ MODELSIM="c:/dev/vunit/acceptance_test/end_to_end_out/modelsim.ini" python -m unittest test_end_to_end.TestVunitEndToEnd

However, if the workaround in modelsim_interface.py is commented out (no additional -modelsimini):
#global env
#set env(MODELSIM) "{modelsimini}"

Then the same cmd-line will work. So it might have something to do with the env(MODELSIM) not being set properly.

If I use a path with spaces and escape them properly in my "manual" environment variable Modelsim i still get the following error:
** Warning: (14) Failed to open "c:/dev/vunit/acceptance_test/end\ to\ end\ out/modelsim.ini" specified by the MODELSIM
environment variable.

All of the tests above have been performed from a cygwin-shell, I might try from a native cmd-shell as well.

tomasnilefrost, 2015-02-06:
Adding os.environ["MODELSIM"] = fix_path(self._modelsim_ini)
to the simulate()-method seems to set the environment variable correcty. 

Still issues with spaces in the path and that solution would break standalone execution of the user.do-files since set env(MODELSIM) has to be removed from the tcl-file.

kraigher, 2015-02-06:
@tomasnilefrost could you try the modelsimini branch?

tomasnilefrost, 2015-02-06:
Doesn't seem to do the trick.

I think that there might be some issues in my version of how the environment variable is treated when it is set from within the tcl script.

kraigher, 2015-02-06:
@tomasnilefrost 
http://www.microsemi.com/document-portal/doc_view/134193-modelsim-me-v10-3a-release-notes

Compatibility Issues with Release 10.3a
User Interface Compatibility
 dvt64978 - (results) Setting the MODELSIM environment variable from within vsim or a
vsim [do] file, would not work correctly; the new setting would not take effect in all
cases. This has been fixed and will now update the simulator and GUI settings
appropriately from the new modelsim.ini file selected by the MODELSIM environment
variable. This change as the potential to effect existing [do] scripts that set this variable.

kraigher, 2015-02-06:
In the above document the bug is listed under "User Interface Defects Repaired in 10.3a".

tomasnilefrost, 2015-02-06:
Hmm, thats wierd.
It's strange that they say that this compatibility issue might affect existing do-scripts that utilize the functionally. I mean if the bug has been fixed in this version.

kraigher, 2015-02-06:
Does "setenv MODELSIM ..." work instead of "set env(MODELSIM) ..." ?

tomasnilefrost, 2015-02-06:
I'll try first thing Monday when I have access to the environment.

kraigher, 2015-02-06:
In summary there are three known bugs in different versions of modelsim in the handling of modelsim.ini:

### Bugs
* **B1** -modelsimini vsim flag not respected in some versions of modelsim, reported in issue #4 

* **B2** -modelsimini vsim flag does not support spaces in path. Adding a correctly space-escaped or {}-wrapped argument simply causes modelsim to complain that the file does not exist. When referring to the file it modelsim uses double {{}}-braces which indicates it does some internal extra escaping. Other vsim flags such as -wlf handle spaces in path using the same method that I have tried on the modelsimini flag. 

* **B3** MODELSIM environment variable not respected when set inside the tcl-script before the vsim command. This bug is reported with microsemi modelsim 10.3a. The release notes of microsemi modelsim 10.3a mentions this bug as fixed though.

### Possible solutions:
* Use both modelsimini flag and environment variable.
   * **Drawbacks**
      * Does not support space in path.
* Always use enivronment variable, only use flag when no spaces in path.
   * **Drawbacks**
      * Does not support space in path for modelsim with **B3** bug.
* Set environment variable in Python before calling modelsim.
   * **Drawbacks**
      * Running the _user.do_ tcl-script when debugging test cases in GUI will not get modelsim.ini variable set correctly.
* Set environment variable in Python before calling modelsim and create python wrapper or _--gui_ flag to launch gui from python.
  * **Drawbacks**
     * None?

tomasnilefrost, 2015-02-09:
It seems that setenv isn't working better at all. Microsemi has had a fair bit of buggy implementations for their fpga flow so I'm not that suprised that there are issues.

I think the solution with a --gui-flag to the ui would be quite beneficial. The user interface could look the same regardless of simulator and it is possible to use wildcards in the ui to quickly select which test bench that the user wants to run using the modelsim gui.

kraigher, 2015-02-09:
I will look into adding a --gui flag.

commit 0bcfe8f2 (2015-02-12): Adds --gui flag. Closes #16
commit 5a59d2b3 (2015-02-19): Merge pull request #20 from tomasnilefrost/wip-fix-environment-variable

Remove tcl flag. Closes #16 for Microsemi 10.3a
closed 2015-02-12 by kraigher