VUnit/vunit #17: Add Aldec support

LarsAsplund, 2015-02-12:


kraigher, 2015-04-08:
**Update:** We are still working with Aldec to get tool provider licenses.

LarsAsplund, 2015-04-08:
I will fill out the requested paperwork today.

kraigher, 2015-05-30:
@LarsAsplund Any update on the tool provider licenses?

LarsAsplund, 2015-06-01:
@kraigher No, no more info. I will get in touch with them after DAC again.

kraigher, 2015-08-24:
@LarsAsplund Any update on the licensing front?

joshrsmith, 2015-08-26:
As a user of Aldec tools, I am also interested in support for Riviera and Active HDL.

After poking around in the code I am going to wager a guess that Riviera support will be extremely similar to Modelsim, since the Riviera interface is almost identical: `vsim, vlib, vcom`, etc all operate in a similar manner.

One difference is Riviera doesn't use *modelsim.ini*, but uses *library.cfg* instead. I believe that by default when *library.cfg* is created it will reference the global *library.cfg* that lives with the installation - to pull in the standard libraries automatically without needing to manually map them.

I was able to run the "user guide" tests on the command line with Riviera after simply hacking out some of the code related to *modelsim.ini* use. Running interactively with the GUI did not work; there are more differences that caused errors I think.

kraigher, 2015-08-27:
@joshrsmith Yes it should be straightforward. The main blocker is that none of the main authors of VUnit has access to any Aldec simulator. We have had talks with Aldec about us obtaining tool provider licenses at just a fraction of the cost of a normal license. We cannot motivate ourselves to pay the full price out of our own pockets. They indicated that we could have such licenses but the process has unfortunately stalled over the summer. Aldec users that are interested in VUnit could help by letting your Aldec representative and Julian Lonsdale (European representative) know that you desire to use VUnit with Aldec products.

An alternative to the main authors getting licenses is that a user would take responsibility for testing the Aldec backend. I could commit to doing the coding work and make sure it is integrated well if a user having access to the tool does the testing. We do have a set of automated acceptance tests that can be run to verify that all VUnit functions work as expected under a given simulator so the testing does not involve many manual steps. Such a solution could work in the short term but in the long term it is important that the main authors are able to independently move forward without being held back by the inability to verify changes against the Aldec backend.

By the way a user already made a [fork](https://github.com/oesvends/vunit/tree/Riviera-Pro) with added Aldec support some time ago. This user got assigned to another project at work and could not work any more on this fork. Thus it is very far behind the VUnit mainline and I would not recommend using it.

LarsAsplund, 2015-09-09:
We now have our licenses!

joshrsmith, 2015-09-09:
Excellent!

kraigher, 2015-09-09:
I have started working on creating an Aldec interface. I have run into some problems though.
Apparently there should be a `breakassertlevel` variable which controls the VHDL severity level at which the tool breaks simulation:
https://www.aldec.com/en/support/resources/documentation/faq/1070
It does not seem to work though. There is no `breakassertlevel` variable defined and setting it to 2 will not make the simulator stop on error. Any Aldec user that has used this?

LarsAsplund, 2015-09-09:
I've opened a case in the customer portal and get back when I get an answer.

kraigher, 2015-09-10:
Active-HDL does not have this "problem" so I guess I will try to get that to work first. There seems to be quite a few differences between Riviera-Pro and Active-HDL with regards to the batch functionality. I have pushed the work in progress to the aldec branch.

LarsAsplund, 2015-09-10:
From Aldec support team:

For Riviera-PRO please use vhdlassert command instead i.e.:

vhdlassert.break error

kraigher, 2015-09-10:
@LarsAsplund Next bug I found in Active-HDL with the write procedure:
```vhdl
use std.textio.all;

entity bug is
end entity;

architecture a of bug is
begin
  p : process
    file fptr : text;
  begin
    file_open(fptr, "file.txt", write_mode);
    write(fptr, string'("hello"));  
    file_close(fptr);
  end process;
end architecture;
```
`file.txt` will contain 4 additional garbage bytes before `hello`. The contents of those 4 prefix bytes seems to depend on the choice of string. With modelsim or GHDL the file contains only 5 bytes for `hello` rather than 9 bytes with Active-HDL.

Could you report this to Aldec?

LarsAsplund, 2015-09-10:
@kraigher Will do

kraigher, 2015-09-10:
I got the artificial acceptance tests to work with Active-HDL now. Had to change -p4 to -p1 since I only have one license and it errors out if you try to use more licenses than you have rather than wait for a license.

LarsAsplund, 2015-09-11:
@kraigher Here is the answer from Aldec

Hi Lars,

In the sent example, the predefined function write() is used.
This function is not defined in std.TEXTIO package.
(see LRM 1076-2008 clause 16.4 - Package TEXTIO, page 269):
``` vhdl
type TEXT is file of STRING; -- A file of variable-length
-- ASCII records.
-- The predefined operations for this type are as follows:
(...)
-- procedure READ (file F: TEXT; VALUE: out STRING);
-- procedure WRITE (file F: TEXT; VALUE: in STRING);
```

According to clause 5.5.2 File operations, page 55: human-readability is not guaranteed by predefined write functions.
If somebody want to write a human-readable file MUST use functions from predefined package: std.TEXTIO.
LRM 1076-2008 clause 5.5.2 - File operations, page 57:
``` console
NOTE 2Predefined package TEXTIO is provided to support formatted human-readable I/O.
```
Our behavior is strict to LRM. The "Funny characters" written to a file.txt file are lengths of written fields coded binary.
They allow to properly use write and read procedure on such file.

Please consider the following code:
``` vhdl
------------------writes2.vhd----------------------
use std.textio.all;

entity writes is
end;

architecture behavior of writes is
  file dump : TEXT;
  file dump2 : TEXT;
begin
  wrtng : process
    variable aa : line;
    variable a : string(1 to 15);
    variable b : natural:=36;
  begin
    FILE_OPEN(dump,"dump_2.txt",WRITE_MODE);
    write(dump,string'( "Item 1; "));
    write(dump,string'( "item 2; "));
    write(dump,string'("item 3; "));
    write(dump,string'( "End Of Line " & LF));
    FILE_CLOSE(dump);
    report "---------------------";
    wait for 1ns;
    FILE_OPEN(dump2,"dump_2.txt",read_mode);
    read(dump2,a,b);
    write(OUTPUT,a);
    read(dump2,a,b);
    write(OUTPUT,a);
    read(dump2,a,b);
    write(OUTPUT,a);
    read(dump2,a,b);
    write(OUTPUT,a);
    wait;
  end process;
end;
```
Other simulators can't correctly read the file (dump_2.txt) that was written by them.
The easiest would be to change code in the following universal way:
``` vhdl
use std.textio.all;

entity bug is
end entity;

architecture a of bug is
begin
  p : process
    file fptr : text;
    variable L_OUT : LINE;
  begin
    file_open(fptr, "file.txt", write_mode);
    write(L_OUT, string'("hello"));
    writeline(fptr,L_OUT); 
    file_close(fptr);
    wait;
  end process;
end architecture;
```

kraigher, 2015-09-11:
Their suggestion is not equivalent since it also writes a newline.

kraigher, 2015-09-11:
I wonder in how many files we assume write works this way.

kraigher, 2015-09-11:
It seems Active-HDL requires that the dependants of a package are re-compiled when a package body has been recompiled.

kraigher, 2015-09-11:
@LarsAsplund My current focus is to get Active-HDL running then I will switch to Riviera-Pro. 
When running the run.py files for our builtin code in vhdl/*/test/run.py there are some errors which is likely due to bugs in Active-HDL. Could you have a look at this and see if it can be worked around and maybe be reported to Aldec? Then I can continue with either `--gui` for Active-HDL or Riviera-Pro. The artificial tests work now with Active-HDL at least.

LarsAsplund, 2015-09-11:
Yes, I can have a look at that.

kraigher, 2015-09-11:
Focus on the 2008 stuff since there is no stop mechanism implemented for aldec that works without 2008 std.env.stop.

kraigher, 2015-09-11:
My impression so far is that the fork that has been made with claimed aldec support could only have worked in a very superficial manner. But we are close now.

joshrsmith, 2015-09-11:
Is the intention to add Active HDL and Riviera support separately? That these would be two independently supported tool chains? I ask because I am seeing references to `aldec` in the source, which isn't very clear which simulator is being referenced -- `activehdl` or `riviera`. It would be similar to reference `modelsim` as `mentor` -- also potentially confusing.

kraigher, 2015-09-11:
@joshrsmith Initially I made the assumption that both Active-HDL and Riviera-Pro could be supported by the same type of interface and that the toolchain could be selected based on what was on your PATH environment variable. After trying out the tools I realize that they have subtle differences that prohibit this and I will probably need two partially overlapping implementations. Thus it will end up with `activehdl` and `rivierapro` as simulators rather than `aldec`. I will correct this mistaken initial assumption very soon.

joshrsmith, 2015-09-11:
This sounds like a good approach. Similar as possible with provisions in place for handling differences where necessary. From my experience they may have some similar features, but they each have their own quirks. Far as I can tell they care for Riviera first since it is the premium product.

kraigher, 2015-09-11:
I ran into some problems implementing a `--gui` flag for active HDL. It seems like the Active-HDL GUI is very different from running through VSimSA. I just want to open the GUI and simulate a top level I compiled on the command line just as I can in ModelSim. Apparently it seems this is not possible. In the GUI everything has to be a "design" or a "workspace". This is inconvenient to say the least. I will investigate if this is more convenient in Riviera-Pro.

joshrsmith, 2015-09-11:
I am not surprised about that with Ahdl. Riviera offers both and line/batch and workspace based flows. I have never tied a pure batched based flow with active Hdl but it is certainly possible with riviera.

kraigher, 2015-09-11:
The batch based works now with Active-HDL. The problem is there is no gui capability with the batch based flow and we would like to have a `--gui` command to open the test preloaded in the gui. Unlike modelsim (and hopefully riviera) you cannot just simulate your batch-flow-compiled design in an open Active-HDL GUI.

kraigher, 2015-09-11:
I am wondering if one could generate a design/workspace also for the batch based flow such that they can be used to also simulate in the GUI.

kraigher, 2015-09-12:
It seems that it is possible to open a batch compiled design in Riviera-Pro GUI and simulate it unlike Active-HDL. The problem is that it will not allow any single stepping through the code. Running `step` just runs the entire simulation until the end. Even though you can navigate the batch compiled libraries in the and find your source files you cannot set breakpoints on them. Maybe I have to abandon the batch flow completely and use workspace/designs all the way. Any Aldec user that can comment on this? In modelsim you could single step and set breakpoints in the GUI using your batch-compiled libraries/files.

kraigher, 2015-09-12:
Adding the `-dbg` switch when compiling enabled me to single step the files with the batch flow in the gui.

kraigher, 2015-09-12:
@LarsAsplund Riviera-Pro triggers the same simulation bugs as Active-HDL. It is probably easier for you to debug using Riviera-Pro with `--gui` flag.

LarsAsplund, 2015-09-12:
Ok. I'll test that this evening. I had a quick look yesterday and found
some issues.  Some are easy to work around but some I rather wouldn't. I
will try to create some simple examples of these issues and discuss them
with Aldec.

LarsAsplund, 2015-09-12:
I also used the debug flag to manually recompile the top-level testbench when I tested yesterday and that seemed to remove some problems so there may be some optimisation issues.

kraigher, 2015-09-12:
The Riviera-Pro interface is turning out well now once I understand the tool. Opening the Hieararchy and Objects panels really helps making it feel familiar to what I am used to.

One remaining task is to investigate if riviera pro also needs extra package body dependencies.
I will create some general acceptance test where a package body is modified and the design is recompiled to flush out simulators which are sensitive to this.

LarsAsplund, 2015-09-13:
@kraigher I had a look at the VHDL tests and found a number of issues. Some easy to work around, some not so much. I also found that the path and directory tests failed because of spaces in the test case names. Maybe you can have a look at that. I will list the things I found to see if someone has a comment. I was able to create most of them in simpler examples so I will send that to Aldec.

LarsAsplund, 2015-09-13:
This is an example of a problem frequently occurring when doing a check where the boolean expression is an equality with a function call on one side. If the same function call is used to create the error message string then the equality fails although it shouldn't. The problem disappears when debug compiling so that's the workaround to use for now.

``` vhdl
entity bug is
end entity bug;

architecture a of bug is
begin
  main: process is
    function echo (
      constant str : string)
      return string is
    begin
      return str;
    end function echo;

    procedure test (
      constant expr : in boolean;
      constant msg  : in string) is
    begin
      if not expr then
        assert false report msg severity error;
      else
        report "Test passed!";
      end if;
    end procedure test;

  begin
	 -- Passes when the echo function isn't called as part of the string
    test(echo("hello") = "hello", "Expected echo(""hello"") = hello");
	-- Fails when echo is part of the string. Passes when debug compiled.
    test(echo("hello") = "hello", "Expected " & echo("hello") & " to equal hello");
	wait;
  end process main;

end architecture a;
```

This will be reported to Aldec

LarsAsplund, 2015-09-13:
Here is example where Active-HDL fails to compile with an error. It fails to associate `foo` in the assignment to `v2` with the function of that name. The other assignments are fine.

``` vhdl
entity bug is
end entity bug;

architecture a of bug is
begin
  main: process is
  	type test_t is (foo);
  	type test2_t is (foo, bar);

    function foo
      return boolean is
    begin
      return true;
    end function foo;

    variable v1 : test_t;
    variable v2 : boolean;
	variable v3 : test2_t;
  begin
    v1 := foo;
    v2 := foo;
    v3 := foo;
    wait;
  end process main;

end architecture a;
```

This will be reported to Aldec

LarsAsplund, 2015-09-13:
Another problem which I haven't been able to recreate in a simple example such as this.

``` vhdl
entity bug is
end entity bug;

architecture a of bug is
begin
  main: process is
    variable v1 : boolean_vector(0 to 3) := (false, false, true, false);
  begin
    v1(1 to v1'right) := v1(0 to v1'right - 1);
    assert v1 = (false, false, false, true);
    wait;
  end process main;

end architecture a;
```

The problem I've seen is that `v1` is all `false` values after the assignment. The workaround is to use a temporary variable in between such that `v1`isn't assigned to itself. This has not been reported to Aldec, Anyone seen this?

LarsAsplund, 2015-09-13:
This is a division problem that affects our ability to print long (>32 bits) unsigned values as integers. In this case the last assert fails.

``` vhdl
library ieee;
use ieee.numeric_std.all;

entity bug is
end entity bug;

architecture a of bug is
begin
  main: process is
    variable v1 : unsigned(115 downto 0) := x"7b283d038f92b837d92f73a87380a"; -- Decimal: 39966790250241720040714860591790090
    variable v2 : unsigned(99 downto 0) := x"08123C108D448411C014EF82F"; -- Decimal: 39966790250241720040714860591
    variable v3 : unsigned(95 downto 0) := x"8123C108D448411C014EF82F"; -- Decimal: 39966790250241720040714860591
  begin
    assert v1/10 = 0x"0C50D2E6C18EAC0595B7F1F73EC01"; -- Decimal: 3996679025024172004071486059179009
    assert v3/10 = 0x"0CE9F9B4153A6CE93354B26B"; -- Decimal: 3996679025024172004071486059
    assert v2/10 = 0x"00CE9F9B4153A6CE93354B26B"; -- Decimal: 3996679025024172004071486059
    wait;
  end process main;

end architecture a;
```

This has been reported to Aldec. I have no workaround for this problem.

LarsAsplund, 2015-09-13:
This is a problem for our com package which means that an encoded character `a` becomes `A` when decoded. This doesn't happen all the time, only in the second assert in this example.

``` vhdl
entity bug is
end entity bug;

architecture a of bug is
  function foo
    return string is
  begin
    return "'a'";
  end function foo;

  function bar
  	return character is
  begin
    return character'value(foo);
  end function bar;

  function bar2
  	return character is
  begin
    return character'value("'a'");
  end function bar2;
begin
  main: process is
  begin
    assert bar2 = 'a' report "Expected a but got " & bar2 & "." severity error;
    assert bar = 'a' report "Expected a but got " & bar & "." severity error;
    wait;
  end process main;

end architecture a;
```

This has been reported to Aldec and I have no workaoround.

LarsAsplund, 2015-09-13:
Another problem with I haven't been able to recreate with a simple example. When I run this code everything is ok but in the com package code the `v1` range created from the range attribute would typically become a null range while the `v2` range is ok. Any idea? 

``` vhdl
entity bug is
end entity bug;

architecture a of bug is
  function bar (
    constant value : positive)
    return positive is
  begin  -- function bar
    return value;
  end function bar;

  function foo (
    constant value : boolean)
    return string is
    variable ret_val1 : string(bar(1) to bar(3));
    variable ret_val2 : string(bar(1) downto bar(3));
  begin
    if value then
      return ret_val1;
    else
      return ret_val2;
    end if;
  end function foo;
begin
  main: process is
    variable v1 : string(foo(true)'range);
    variable v2 : string(foo(true)'left to foo(true)'right);
  begin
    assert v1'left = 1;
    assert v1'right = 3;
    assert v2'left = 1;
    assert v2'right = 3;
    wait;
  end process main;

end architecture a;
```

This has not been reported to Aldec. There are workarounds but they mean that I have to change in many places.

LarsAsplund, 2015-09-14:
@kraigher 9ef20e36a6a5259fe3f0d33e7671a3b5a067c960 seems to break vhdl/logging/run.py. Could you have a look?

kraigher, 2015-09-15:
@LarsAsplund I can run it without problems on `master` using modelsim/ghdl? What is broken for you?

LarsAsplund, 2015-09-15:
@kraigher This is what I'm seeing. I was rather quick making that assumption. I just looked at the last method being and didn't make any investigation of my own.

``` console
d:\Programming\github\vunit\vhdl\logging>python run.py --clean
Traceback (most recent call last):
  File "D:\Programming\github\vunit\vunit\ui.py", line 324, in main
    all_ok = self._main()
  File "D:\Programming\github\vunit\vunit\ui.py", line 353, in _main
    self._compile(simulator_if)
  File "D:\Programming\github\vunit\vunit\ui.py", line 439, in _compile
    simulator_if.compile_project(self._project, self._vhdl_standard)
  File "D:\Programming\github\vunit\vunit\activehdl_interface.py", line 84, in compile_project
    for source_file in project.get_files_in_compile_order():
  File "D:\Programming\github\vunit\vunit\project.py", line 256, in get_files_in_compile_order
    dependency_graph = self._create_dependency_graph()
  File "D:\Programming\github\vunit\vunit\project.py", line 236, in _create_dependency_graph
    add_dependencies(self._find_other_design_unit_dependencies, vhdl_files)
  File "D:\Programming\github\vunit\vunit\project.py", line 226, in add_dependencies
    for dependency in dependency_function(source_file):
  File "D:\Programming\github\vunit\vunit\project.py", line 158, in _find_other_design_unit_dependencies
    yield library.get_package_body(primary_unit.name).source_file
  File "D:\Programming\github\vunit\vunit\project.py", line 450, in get_package_body
    return self._package_bodies[name]
KeyError: 'lang_mock_types_pkg'
```

kraigher, 2015-09-15:
@LarsAsplund Are you missing some file? Do you have a clean repository? I would have seen the same problem here as well but I do not.

kraigher, 2015-09-15:
@LarsAsplund I think the problem is that `lang_mock_types_pkg` does not have a package body. But you only notice this when running with Riviera Pro since with modelsim package user to package body depdendencies are disabled.

kraigher, 2015-09-15:
@LarsAsplund I have fixed the problem now. By the way I saw that you pushed stuff to the `aldec` branch. Since I often force push history changing stuff to that branch you might not always be able to do `pull` but rather have to do interactive rebase.

LarsAsplund, 2015-09-15:
@kraigher I'll try your fix later today.

LarsAsplund, 2015-09-15:
@kraigher The fix works.

LarsAsplund, 2015-09-15:
@kraigher The problem with running test cases with names containing spaces is solved if the file name in `do -tcl <file name>` is enclosed in double double quotes (""). I just tried it so that solution is not based on any documentation I found.

kraigher, 2015-09-15:
@LarsAsplund What problems are you seeing with space in test name? I have run the `path` and `dictionary` tests with Riviera Pro and they all pass.

LarsAsplund, 2015-09-15:
@kraigher The fix I did was under Acitve-HDL but I see the same problem when switching to Riviera-PRO

``` console
d:\Programming\github\vunit\vhdl\path>python run.py *sep*
Starting lib.tb_path.Verify that a separator ending the first path is ignored

VHDL/Verilog/EDIF/SystemC Simulator build 10.2.3312.5682

(c) 1997-2015 Aldec, Inc. All rights reserved.

License Number 0

VSIMSA: Configuration files: `d:\Programming\github\vunit\vhdl\path\vunit_out\rivierapro\library.cfg', `C:\Aldec\Active-HDL-10.2\BIN\vsimsa.cfg'
 Welcome to VSIMSA!
 This message was printed from `startup.do' macro file.

log "d:\Programming\github\vunit\vhdl\path\vunit_out\tests\lib.tb_path.Verify that a separator ending the first path is ignored\asim\transcript"
log -assert "d:\Programming\github\vunit\vhdl\path\vunit_out\tests\lib.tb_path.Verify that a separator ending the first path is ignored\asim\transcript"
Cannot open file `source d:/Programming/github/vunit/vhdl/path/vunit_out/tests/lib.tb_path.Verify'.
fail (P=0 S=0 F=1 T=1) lib.tb_path.Verify that a separator ending the first path is ignored (1.6 seconds)

==== Summary ================================================================================
fail lib.tb_path.Verify that a separator ending the first path is ignored (1.6 seconds)
=============================================================================================
pass 0 of 1
fail 1 of 1
=============================================================================================
Total time was 1.6 seconds
Elapsed time was 1.7 seconds
=============================================================================================
Some failed!

d:\Programming\github\vunit\vhdl\path>
```

kraigher, 2015-09-15:
@LarsAsplund I have put Active-HDL on ice and have focused on Riviera-Pro. The Active-HDL interface is not anything I would recommend you using at the moment. My suggestion is that we focus on Riviera-Pro only for the time being. Are you sure you are using Riviera-Pro the printout you added looks like it was from Active-HDL. You need to ensure that Riviera-Pro is first on your PATH otherwise you risk that Active-HDL binaries are used even with rivierapro_interface.py. We need to make the simulator tool chain detection mechanism more advanced to differentiate them since the executables are named the same but this is not yet done.

kraigher, 2015-09-15:
When I trick the riverapro_interface.py to use Active-HDL binaries like this I can reproduce your errors.
```
PATH=/C/Aldec/Active-HDL-10.2-x64/BIN/:$PATH PYTHONPATH=. VUNIT_SIMULATOR=rivierapro python vhdl/dictionary/run.py -v
Starting lib.tb_dictionary.Test that an empty frozen dictionary has zero length

VHDL/Verilog/EDIF/SystemC Simulator build 10.2.3312.5682

(c) 1997-2015 Aldec, Inc. All rights reserved.

License Number 0

VSIMSA: Configuration files: `c:\Users\kraigher\repo\vunit\vunit_out\rivierapro\library.cfg', `c:\Aldec\Active-HDL-10.2-x64\BIN\vsimsa.cfg'
 Welcome to VSIMSA!
 This message was printed from `startup.do' macro file.

log "c:\Users\kraigher\repo\vunit\vunit_out\tests\lib.tb_dictionary.Test that an empty frozen dictionary has zero length\asim\transcript"
log -assert "c:\Users\kraigher\repo\vunit\vunit_out\tests\lib.tb_dictionary.Test that an empty frozen dictionary has zero length\asim\transcript"
Cannot open file `source c:/Users/kraigher/repo/vunit/vunit_out/tests/lib.tb_dictionary.Test'.
```

LarsAsplund, 2015-09-15:
@kraigher Yes, you're right. My path wasn't updated. I've been trying AHDL and RPRO, debug and no debug back and forth but I can stick with RPRO. Switching between debug and no debug is more interesting to get the VHDL code working. I still have issues with the run test cases that works with debug but not with no debug so I will work on that.

kraigher, 2015-09-15:
Yes lets stick with Riviera-Pro until it is done.

LarsAsplund, 2015-09-16:
Managed to create a simple example of one of the bugs I mentioned in a previous post.

``` vhdl
entity bug is
end entity bug;

architecture a of bug is
  function get_range
    return bit_vector is
    variable ret_val : bit_vector(1 to 17);
  begin
      return ret_val;
  end function get_range;
  
  function foo
    return string is
    variable ret_val : string(get_range'range);
  begin
    return ret_val;
  end;

begin
  main: process is
  begin
	report foo;
    wait;
  end process main;

end architecture a;
```

This will fail with

``` console
# RUNTIME: Fatal Error: RUNTIME_0047 bug6.vhd (14): Index 0 out of range (1 to 2147483647).
# KERNEL: Time: 0 ps,  Iteration: 0,  Instance: /bug,  Process: main.
# KERNEL: Stopped at time 0 ps + 0.
# VSIM: Error: Fatal error occurred during simulation.
```

where line 14 is the string variable declaration. 

I use this type of code in the com package where the `get_range` function takes an encoded array of some type as input (everything is encoded to strings so the input is a string) and returns an array of a fix type (bit_vector in the example) but with the same range as the encoded array. I use the range of the returned array to declare a variable of the array type I want to decode (string in the example). Once I have an array of the correct type and range I can assign that the decoded values. The idea is that I don't have to assume any max length or if the array has an ascending or descending range. The point of keeping the original range through the encode/decode process and not just the length is that some array types like `fixed` cares about the exact range, not only the length.

I've reported this to Aldec. Suggestions for workarounds are appreciated.

[25 more replies]

closed 2015-09-22 by kraigher