VUnit/vunit #18: Sigasi Integration

LarsAsplund, 2015-02-12:
Agreed with @philippefaes to setup Sigasi integration. This issue will be the placeholder for discussing and implementing initial ambition

philippefaes, 2015-03-20:
As a first step, we want users to get easy access to the vunit_lib files. Ideally, they would just include one directory without the need to include/exclude too many other files. The file structure now takes a few minutes of extra work. Sadly, this causes user drop-off. Especially if users have their own compilation scripts, this might be important.

Is there any chance we can change the file structure to something like this:

    vunit/vhdl # users need to compile this directory in the vunit_lib library
    vunit/vhdl/93 # users need this if they use VHDL 93
    vunit/vhdl/200x # users need this if they use VHDL 2002 or 2008
    vunit/vhdl/mock # not needed for users
    vunit/vhdl/test  # not needed for users

kraigher, 2015-04-08:
@LarsAsplund what do you think about the suggestion made by @philippefaes ? I think it sounds good.

LarsAsplund, 2015-04-08:
@philippefaes @kraigher Yes, I'm in favour of this as well. We would need a 2008 directory as well for the array package (or make it a 200x).

LarsAsplund, 2015-06-18:
@philippefaes @kraigher 

Today the structure is

```
vhdl
vhdl/<module>
vhdl/<module>/src
vhdl/<module>/src/<src_file>
vhdl/<module>/test
vhdl/<module>/test/<test_file>
```

If we should be clear about where the user source files and the developer test files are we should move src and test to the top, then we have the VHDL version libraries to easily identify the subset to compile, and finally we have the module directories to keep things modular by function. Mocks should be moved to test, shouldn't have been under src in the first place.

```
vhdl
vhdl/<src or test>
vhdl/<src or test>/<93, 200x, 2008, or common>
vhdl/<src or test>/<93, 200x, 2008, or common>/<module>
vhdl/<src or test>/<93, 200x, 2008, or common>/<module>/<src_or_test_file>

Ok?

kraigher, 2015-06-23:
@LarsAsplund seems ok. I would prefer if there were no compile-time mocking though. See issue: #52 52

philippefaes, 2015-06-24:
Sounds good.

LarsAsplund, 2015-08-29:
@philippefaes I've created a [branch](https://github.com/LarsAsplund/vunit/tree/restructure-vhdl-directories/vhdl) with the new structure. Still looking good?

kraigher, 2015-08-31:
I would prefer:
`vhdl/<module>/<src or test>/<93, 200x, 2008, or common>/`

instead of:
`vhdl/<src or test>/<93, 200x, 2008, or common>/<module>`

In that way independent modules are kept together under the same folder root and are not interleaved with other unrelated modules in some kind of global structure. Making this transformation would still solve the problem that the non-transformed structure aimed to solve.

philippefaes, 2015-08-31:
@kraigher either way works for me, as long as the dir structure dictates in which library to compile things. I see your point about interleaving modules, though.

@LarsAsplund I seems that the directory names do not map 1-on-1 to library names. It would be highly appreciated if that were the case. 

For example, I see library clauses that use `lib`, `shuffler_lib`, `tb_uart_com`, `SynthWorks` and it is not clear at first sight where they should be mapped.

LarsAsplund, 2015-08-31:
@kraigher @philippefaes 

For libraries Iike VUnit I prefer using `work` for references to itself such that the user can decide a suitable name for the library, see http://stackoverflow.com/questions/31972810/how-to-make-the-library-work-work/31998749#31998749. For that reason I rather not have a directory structure indicating otherwise. We use `vunit_lib` as the default name (`vunit` became reserved in VHDL 2008) but that can be changed in the script (unfortenately you also have to rewrite the provided `vunit_context` since contexts must use hardcoded library names).

So far I've only rearranged in the VUnit library files and not in the examples. Again I would like to avoid another level of directories for this. Maybe we can choose more obvious library names or add comments to the example testbenches. Any naming convention suggestion @philippefaes?

The `SynthWorks` library name is only used by the OSVVM demo as far as I can see. It's there unmodified because we only redistribute OSVVM. We use `osvvm` as default when adding OSVVM from the run script but it can be changed to something else.

kraigher, 2015-09-16:
I have added a small [recipe](https://github.com/LarsAsplund/vunit/blob/master/user_guide.md#creating-sigasi-projects) of how to create a Sigasi project from a VUnit project. For me this is the ideal integration and what I use at work. The script uses [SigasiProjectCreator](https://github.com/sigasi/SigasiProjectCreator) which is a Python library provided by Sigasi to generate a project.

ttobsen, 2016-07-28:
@kraigher @LarsAsplund 

What's the current status on this topic? Has VUnit now an usable Sigasi integration or are there still work in progress? Unfortunately the above recipe link is not working. :(

LarsAsplund, 2016-08-03:
@ttobsen I've been meaning to take care of this or close the issue and now you gave me a reason. Let's start with the deleted recipe. I do not have Sigasi installed on this computer so I cannot test it but from what I can tell the VUnit parts should work except for the `get_project_compile_order` method which has changed to `get_compile_order`. Can you see if it still works? Is this integration what you're looking for? @kraigher Why was the recipe deleted?

## Creating Sigasi Projects
The official Sigasi [SigasiProjectCreator](https://github.com/sigasi/SigasiProjectCreator) utility can be used to generate a Sigasi project from a VUnit project using the following script. The script will create virtual folders for each VHDL library containing the files of the library.
```python
from SigasiProjectCreator import SigasiProjectCreator
from os.path import relpath, exists, basename
import os
from shutil import rmtree

def create_sigasi_project(prj, name, project_path):
    """
    Create Sigasi project from VUnit object prj with name into project_path
    """
    creator = SigasiProjectCreator(name, 2008)
    
    for source_file in prj.get_project_compile_order():
        file_name = source_file.name
        library_name = source_file.library.name
        virtual_name = library_name + "/" + basename(file_name)
        creator.add_link(virtual_name, file_name)
        creator.add_mapping(virtual_name, library_name)

    if not exists(project_path):
      os.makedirs(project_path)

    creator.write(project_path)

```

ttobsen, 2016-08-16:
I tried to follow your instructions. But I'm no Python expert, so it's possible that I'm not able to follow your instructions and answer your questions properly.

What I did;

1.) Created new VHDL project in Eclipse using Sigasi Plugin.
2.) Copied the SigasiProjectCreator in my project.
3.) Created a run.py Script and added your mentioned code snipped.
4.) Changed `get_project_compile_order` to `get_compile_order`.
5.) Called `create_sigasi_project` in run.py using `create_sigasi_project("VUnit_Sigasi", "VUnit_Sigasi", ".")`
6.) Started run.py

But then I got Python error messages:

`  File "./run.py", line 26, in create_sigasi_project
    for source_file in prj.get_project_compile_order():
AttributeError: 'str' object has no attribute 'get_project_compile_order'`

Did I tested what you expected or am I on a total wrong way?

LarsAsplund, 2016-08-16:
@ttobsen Did you really do step 4? It is still complaining about `get_project_compile_order()`. Maybe you can provide the script?

ttobsen, 2016-08-17:
Yes I did, but I copied the old error message. the correct message is:

`Traceback (most recent call last):
  File "./run.py", line 49, in <module>
    create_sigasi_project("VUnit_Sigasi", "VUnit_Sigasi", "..")
  File "./run.py", line 26, in create_sigasi_project
    for source_file in prj.get_compile_order():
AttributeError: 'str' object has no attribute 'get_compile_order'`

Here is the runscript I use:

[run.txt](https://github.com/VUnit/vunit/files/422228/run.txt)

I think the best solution for an easy Sigasi integration would be what @philippefaes mentioned on his first post:

https://github.com/VUnit/vunit/issues/18#issuecomment-84033332

It also would be a much cleaner folder structure for the repository.

LarsAsplund, 2016-08-17:
@ttobsen Change your script to call `create_sigasi_project` just before the call to `ui.main()`. The first argument of the call should be `ui` and not a string.

tivaliy, 2017-01-05:
@LarsAsplund Are you going to make a video about how to integrate Sigasi and VUnit? In my case I have some issues... Here is my source code:

```
#!/usr/bin/env python

from os.path import join, dirname
from vunit import VUnit

from SigasiProjectCreator.SigasiProjectCreator import SigasiProjectCreator
from os.path import exists, basename
import os

def create_sigasi_project(prj, name, project_path):
    """
    Create Sigasi project from VUnit object prj with name into project_path
    """
    creator = SigasiProjectCreator(name, 2008)

    for source_file in prj.get_compile_order():
        file_name = source_file.name
        library_name = source_file.library.name
        virtual_name = library_name + "/" + basename(file_name)
        creator.add_link(virtual_name, file_name)
        creator.add_mapping(virtual_name, library_name)

    if not exists(project_path):
        os.makedirs(project_path)

    creator.write(project_path)

root = dirname(__file__)

ui = VUnit.from_argv()
lib = ui.add_library("lib")
lib.add_source_files(join(root, "*.vhd"))
create_sigasi_project(ui, "VUnit_Sigasi", os.getcwd())
ui.main()
```
This script creates two files: '.project' and '.library_mapping.xml', but can't open/import then this project in Sigasi Studio Creator? Any ideas?

kraigher, 2017-01-05:
@tivaliy A sigasi project is a folder containing the `.project` and `.library_mapping.xml` files. Thus your `dff_project` folder becomes a sigasi project in your solution as presented above. You could import that project using the import project dialog. If you think this is ugly, which I would, you could pass `join(os.getcwd(), "sigasi_project")` into `create_sigasi_project` instead and get a subfolder called `sigasi_project` containing the project.

tivaliy, 2017-01-06:
@kraigher, tanks. In my case the problem is that if I create sigasi project in a such way then I can't use external compiler (e.g., modelsim) in sigasi (it doesn't work), also it's not possible to set some part of the project as a top level and run modelsim from sigasi. Can you also give me an idea (best practice from your point of view) about how to organize project structure in this case?

project_name/
├── run.py
├── SigasiProjectCreator
└── sigasi_project/
        ├── .project
        ├── .library_mapping.xml
        └── src/
                  ├── *.vhd
                  └── tests/
                          └── tb_*.vhd

kraigher, 2017-01-06:
@tivaliy I would not put the source files inside the `sigasi_project` folder since it is a generated folder. The sigasi project is never checked into version control but is just an artifact that is generated by a script.

I do not know why you cannot use modelsim as an external compiler in Sigasi when you have used SigasiProjectCreator to create the project. I myself never use an external compiler in Sigasi since i use VUnit to compile and run my simulations.

Setting a single "top level" in sigasi and using it to launch a simulation makes no sense when having 10-100 of test benches which each can be a top level for simulation like you can do with VUnit. VUnit also supports launching multiple runs of the same test bench top level with different combinations of generics.

A project structure I typically use is:
```
project_name/
|-- run.py
|-- sigasi_project # This folder is generated by a script and not checked into version control
|-- src
     |--- module1
            |---- file.vhd
            |---- tb_file.vhd
     |--- module2
```

`module1` would be compiled into `module1_lib` to get some name entity name conflict protection between modules. My run.py script just uses `glob` to automatically add all modules and files in the `src` folder.

tivaliy, 2017-01-06:
@kraigher , thanks for the tips.
Can you give me some more help? So, now, after running a script, I have the following project structure:

project_name/
├── run.py
├── sigasi_project
│&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;├── .library_mapping.xml
│&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;└── .project
├── src
│&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;├── d_ff.vhd
│&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;└── tb_dff.vhd
└── vunit_out
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;├── modelsim
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;├── preprocessed
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;├── project_database
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;└── test_output 

Then, what kind of import should I use:
1. 'File -> Import -> [general] Existing Projects into Workspace'
2. 'File -> Import -> [sigasi] Import a VHDL project'

In the first case I have the following workspace structure:
VUnit_sigasi/
|----- Common Libraries
|----- dff_lib
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;|---- d_ff.vhd
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;|---- tb_dff.vhd
|----- vunit_lib
Then, how can I get access to my source to make changes, add more tb and so on?

kraigher, 2017-01-06:
Yes you can use both method 1 and 2.

What do you mean exactly with "getting access to my source to make changes"? You can edit your vhdl source files in Sigasi after you open your project.

If you add more files you have to re-generate the project. The point of generating a sigasi project in the first place is that the project itself is not the original source of information of what files and libraries you have. In this use case VUnit defines what source files and libraries you have and the sigasi project is generated from that. This is a specific instance of a general principle that says that any time the original source of information is changed all output artifacts that was generated from that original source of information must be re-generated.

This is just one way of doing it that I myself prefer. One could also imagine the other way around where VUnit would use a sigasi project as the original source of information on what files and libraries there is. Then the sigasi project would be checked into version control.

The reason I prefer my way of doing it is that I like the files and libraries to be defined just by the way they are physically located in the folder structure without me having to individually name each file in some list in a project.

As a side note the visual representation in a sigasi project generated in this example is "virtual" in the sense that it is organized around the library names rather than the physical folder locations of the files. If you want it to be organized around the folder location instead you can change the argument to the `add_link` method to be the relative path of the file compared to the project root.

tivaliy, 2017-01-06:
@kraigher thanks for you explanation.
Actually, I always use Sigasi as EDA Tool for VHDL-project entry. And it getting me a little bit confusing what is the design flow for VUnit + Sigasi. I thought that Sigasi is used to create project that contains VHDL files, testbench modules. And only after that I can involve VUnit.
As far I understand you use some other tools for project entry, then copy those source and tb files to 'src' (with respective modules) and only after that you execute 'run.py'. Am I right?

Thanks for your tips about ```add_link``` method. I change path to make it absolute. Now I can modify files from my 'diff_lib' and even run them in Modelsim. But still confusing about design flow. Because I have to modify project source files, then copy them to 'src' folder and then re-generate project.

kraigher, 2017-01-06:
@tivaliy You never need to copy any vhdl files into the sigasi project folder. The sigasi project never "contains" the files it just references them.

I use Sigasi to navigate and edit the vhdl code. I use a Python script to generate a Sigasi project automatically from the project structure rules I have set up. The generated Sigasi project will reference the files that are in my project folder that is commited to version control. If I add a new file or folder I just do that in the File Explorer, on the command line or in Sigasi. When I add a new file I either re-generate the Sigasi project or simply add the file to the Sigasi project manually without re-generating. From a clean code repository I can always generate a consistent Sigasi project.

tivaliy, 2017-01-06:
@kraigher , thanks, got it.

noasic, 2017-02-07:
What's the current status on this? What's the recommended way to add the VUnit packages to a Sigasi project?

kraigher, 2017-02-07:
@noasic Use [SigasiProjectCreator](https://github.com/sigasi/SigasiProjectCreator) library to create the project and VUnit API to get the list of files to add to it. That is how I do it at work for a long time. I do not submit the sigasi project to version control only the script that creates it. See the example code above.

noasic, 2017-02-08:
@kraigher thanks

igmar, 2017-04-03:
It would be nice to get this on the road : I'm facing issues setting up vunit within Sigasi.

kraigher, 2017-04-03:
@igmar See my answer to @noasic above. We recommend using SigasiProjectCreator to programatically create a Sigasi Project containing the VUnit files and other user specific project files.

igmar, 2017-04-03:
@kraigher I tried, but it's pretty much black magic to me.

LarsAsplund, 2017-04-04:
Sigasi has plans to integrate VUnit with Sigasi Studio to get the same kind of experience you have when running other unit testing frameworks like JUnit and PyUnit from within Eclipse. 

I've informed Sigasi (@heeckhau) about this issue such that they can provide any progress they like to share.

LarsAsplund, 2017-09-28:
With the release Sigasi Studio 3.6 I think we've covered the original intent of this issue. For more information see http://insights.sigasi.com/tech/vunit-quickfix.html

closed 2017-09-28 by LarsAsplund