VUnit/vunit #24: Add GHDL simulator support

kraigher, 2015-02-19:
We would like to add support for GHDL. There are some blockers though:

1. GHDL does not allow setting top level generics during elaboration from the command line. We rely on this feature quite a lot to pass information from the Python-runner to the test bench. Just being able to set 'string' generics would cover our basic needs. Having support for integer, boolean and real would be nice though since it is used by some of our examples.

2. GHDL does not support breaking simulation when a signal changes value to 'true'. We use the 'exit_without_errors' boolean to break the simulation in the test_runner_cleanup function.

We could try to contact the author of GHDL and see if the above features could be implemented so that we can add GHDL support to VUnit.

LarsAsplund, 2015-02-20:
I've added a [ticket](https://sourceforge.net/p/ghdl-updates/tickets/37/) to the GHDL project.

kraigher, 2015-02-21:
When they fix 1. as they promised I will have a go at adding GHDL support.

kraigher, 2015-03-09:
I see they have fixed 1 on their development branch. Now we just have until there is a binary release.

kraigher, 2015-05-09:
I have managed to build ghdl from the latest commit using the LLVM backed and started ironing out the problems of supporting it. I have found a bug in ghdl when compiling the VUnit code: http://sourceforge.net/p/ghdl-updates/tickets/41/

kraigher, 2015-05-10:
I have successfully run a modified user_guide example using VUnit and GHDL now with just a few work-arrounds.

kraigher, 2015-05-11:
The test_run.py acceptance test works with GHDL now.

kraigher, 2015-05-11:
I have pushed my work in progress to the `ghdl` branch. It only works with GHDL built from the latest source. I have tested it using the LLVM backed on Ubuntu 14.04 LTS. There are some issues and work-arrounds still but many things work such as uart example, array example, osvvm_integration example, generate_tests example etc. It will only work with VHDL 2008 since for the other standards we have not implemented any stop mechanism for GHDL. Even though the VHDL-2008 supports seems quite large there is no support for contexts so I removed those. Also OSVVM 2015.03 has two LRM violations which GHDL errors on so I temporarily fixed those in this branch. I have posted the violations to the OSVVM forum and Jim Lewis has already promised to fix one of them in the next OSVVM release. 

@LarsAsplund There are some LRM violations in our code especially the mocked code that GHDL errors out on. I will try to gather a list of things we need to fix. One common issue seems to be that unlike modelsim GHDL will error out on having shared variables of non-protected type for std >= 2002.

I will keep grinding at this by building the latest GHDL from source and report tickes and hopefully when they release their next official binary release we will have VUnit support for it.

LarsAsplund, 2015-05-11:
@kraigher Nice progress! I'll take a look at the non-protected shared variables to see if they can be easily removed. Have you spoken to the GHDL team about context support? It's very convenient to have that.

kraigher, 2015-05-11:
@LarsAsplund Yes I have submitted a feature request for it. You can look at their [Tickets](https://sourceforge.net/p/ghdl-updates/tickets/) many of the latest ones are mine. Some of the recently closed tickets are bugs I found in ghdl on using subprogram aliases and ambiguity checks which they quickly fixed. At first our code would not even compile on ghdl but now it does.

I am actually quite impressed by the latest GHDL development branch, it has basic vhdl 2008 support and with the LLVM backend it runs fast and is not hard to build from source unlike the gcc backend.

kraigher, 2015-05-11:
They just fixed https://sourceforge.net/p/ghdl-updates/tickets/47/ so that generics without default values will not cause error even though they were specified by the `-g`flag. I have force pushed the `ghdl` branch where I have removed the work arround of having default values on all generics in the examples.

kraigher, 2015-05-11:
They just fixed https://sourceforge.net/p/ghdl-updates/tickets/51/ which prevented the output binary generated by ghdl from being an absolute path. I have removed our work around for it and force pushed the `ghdl` branch.

kraigher, 2015-05-12:
I have rebased the `ghdl` branch on master after com was added and applied work the required workarounds. It crashes when with a failing assert of msg_codecs_pkg.vhd and I will investigate why and send a reproducer to the GHDL team. They are usually quick to fix such things.

kraigher, 2015-05-12:
I have found the problem with msg_codecs_pkg.vhd and condensed it into a ticket:
https://sourceforge.net/p/ghdl-updates/tickets/55/

kraigher, 2015-05-12:
They have now added support for context clauses so I have force pushed an updated `ghdl` branch with those workarounds removed.

kraigher, 2015-05-12:
They have fixed https://sourceforge.net/p/ghdl-updates/tickets/55/ which was caused by msg_codecs_pkg.vhd. Now there is a problem with scoreboard.vhd which I will investigate.

kraigher, 2015-05-13:
I have located the bug which causes scoreboard.vhd to fail:
https://sourceforge.net/p/ghdl-updates/tickets/59/
They have now fixed it.

They have also used to have an error when a the timestamp of an already analyzed file had changed.
This caused problems with git since it will often change timestamps as well as with preprocessed and generated codec files. They now check the sha1 hash of the file instead of the timestamp:
https://sourceforge.net/p/ghdl-updates/tickets/48/

With these fixes GHDL runs the com example without problems.

LarsAsplund, 2015-05-13:
How many of the acceptance tests have successfully run with GHDL. What's left?

kraigher, 2015-05-13:
@LarsAsplund Some work at least. I need to refactor the acceptance test code to handle multiple simulators. Also at least initially I think we can only support GHDL with VHDL-2008 since there is no stop mechanism available otherwise. In modelsim we use the wait on test_runner_exit to not rely on stop. A future enhancement might be to add a report with severity failure as a possibility for GHDL < 2008 but there is really no good reason for anyone to use a 2008 compatible simulator without using 2008. I will be away on travel until Sunday so I will have a bit more limited time working at this. But I will gather a list of what  fails after I have refactored the acceptance test code.

LarsAsplund, 2015-05-15:
@kraigher I push an update to master which removes unprotected shared variables

kraigher, 2015-05-15:
@LarsAsplund They have added the option of turning shared variable error into a warning with `-frelaxed-rules` yesterday. Tank you anyway. By the way I think we should seriously consider dropping support for 93 and even 2002, it costs alot of extra code especially for 93 and it makes the interface worse than it could have been. If we cannot support Vivado xsim then there are no simulators which we are about to support which do not have some subset of 2008 and there is no reason to not use 2008 when simulating.

kraigher, 2015-05-15:
@LarsAsplund One of the acceptance test to fail is `tb_logging.vhd` where the exact contents of log messages are compared. Log messages with times in them fail due to ModelSim time`image formatting with `ps` whereas ghdl formats with `fs`. time'image will return values with the unit defined by the current simulator resolution which we have set to ps in Modelsim. Maybe we have to set it to fs in Modelsim asvwell to be consistent unless we can make our own time to string function.

LarsAsplund, 2015-05-17:
@kraigher Yes, 93 support is adding quite a lot of work. I'm not too concerned with the lack of xsim support. I know that Xilinx has been talking about supporting VHDL 2008 as well, not sure when though. There's also a question about corporate standards. I know some major companies have restrictions but it's unclear to me whether those restrictions apply to testbenches as well. I will try to find out how widespread such restrictions are.

LarsAsplund, 2015-05-17:
@kraigher I'll have a look at the logging testbench. I guess I could allow any unit as long as the string represent the expected time.

kraigher, 2015-05-18:
@LarsAsplund Any restriction on using VHDL-2008 in test benches would fall apart quite quickly if questioned. There is just no good reason. There might be some good reasons to use 93 only for the code intended to be used by the synthesis tool but even this it is becoming less so. I think it costs us a lot to support 93 for no good reason.

kraigher, 2015-05-18:
@LarsAsplund The com example fails due to the impure `split_group` procedure from `com_debug_codec_builder.vhd` being called in pure decode functions. The `split_group` function is impure since it uses `check` which uses a shared variable.

LarsAsplund, 2015-05-19:
@kraigher I removed the checks.

kraigher, 2015-05-19:
The com example runs with GHDL now.

kraigher, 2015-05-19:
tb_logging.vhd runs with both GHDL and ModelSim now as I use `time'image` to create the reference strings instead of hardcoding it making it more independent of simulator resolution and `time'image` format.

kraigher, 2015-05-19:
@LarsAsplund com had some problems with reading `out` parameters. I fixed it in 6d1a9ef532db62723945a260a661f2bc22278e92 on the `ghdl` branch. What do you think?

kraigher, 2015-05-19:
@LarsAsplund `tb_check_stable` failed with GHDL. It seems it dependend on process execution order. The log count was verified in the same delta-cycle as it was set and on GHDL the proceses executed in different order from modelsim. I have fixed this in the `ghdl` branch by commit 11541981d264d6cd13d09453cf89417a131fc45c. Seems reasonable, or could it hide some problem?

kraigher, 2015-05-19:
Now all acceptance test pass except `test_com_debug_vhdl_2008` and `test_com_vhdl_2008`. I have skipped all tests on 2002 and 93 standards since the current GHDL interface does not support them. I have also skipped the `test_end_to_end.py` since most test there depend on the `vunit_finished` signal to stop which GHDL cannot support. I will refactor tests using `vunit_finished` signal to instead call a `vunit_stop` procedure which can do VHDL-2008 stop when using ghdl.

The com tests fails have been condensed to the following two tickets:
https://sourceforge.net/p/ghdl-updates/tickets/73/
https://sourceforge.net/p/ghdl-updates/tickets/72/

LarsAsplund, 2015-05-19:
@kraigher I haven't looked at the details but where is a variable like receipt read? Logically it's an output and it would be better if the interface reflect that. Something that can be solved with local temporary variables?

kraigher, 2015-05-19:
@LarsAsplund com.vhd:652-658 for instance: 
```vhdl
  if receipt.status = ok then
    messenger.send(message.sender, receiver, message.request_id, message.payload.all, receipt);
  end if;

  if receipt.status = ok then
    notify(net);
  end if;
```

Could also be solved with a local variable.

LarsAsplund, 2015-05-19:
@kraigher Ok, in that case I would prefer the local variable. Another question regarding ticket 72 and 73. Have I used

proc(s(4 to 15) => "Hello world!");

in com?

proc(s => (4 to 15 => "Hello World"));

seems more clear to me.

kraigher, 2015-05-19:
It looks like this in tb_com_codecs.vhd:
```
proc((4 to 15 => "Hello World"));
```

The other cases I found while investigating and so I reported them as well.

kraigher, 2015-05-19:
@LarsAsplund I fixed the com reading output violations by using local variables and pushed to master.

kraigher, 2015-05-19:
@LarsAsplund On line 53 of com_types.vhd there is a comment about modelsim not being able to handle `time'high`. I tried it on my machine and it works. The problem is `1000 hr` will overflow in GHDL. I printed `time'high` in modelsim and it was `9223372036854775807 ps` in GHDL it was `9223372036854775807 fs`. It would be a nice solution. Otherwise a negative time can be used as well to denote no timeout.

LarsAsplund, 2015-05-19:
@kraigher I can't remember exactly why I had a problem with time'high, it was some time ago. Maybe there was another problem.... Let me try again later on tonight. If I can't see any problems I will change back. If there is a problem a negative time is a better solution.

LarsAsplund, 2015-05-19:
@kraigher I made a quick test. My com example gets a fatal error when I use time'high. Windows + Modelsim Altera Edition 10.1d. Get back when I have some more details but here is the output

```console
# ** Fatal: (SIGSEGV) Bad handle or reference.
#    Time: 0 ps  Iteration: 6  Process: /tb_card_shuffler/scoreboard/main File:
D:/Programming/github/vunit/examples/com/vunit_out/preprocessed/tb_shuffler_lib/
scoreboard.vhd
# Fatal error in Process main at D:/Programming/github/vunit/examples/com/vunit_
out/preprocessed/tb_shuffler_lib/scoreboard.vhd line 61
#
# HDL call sequence:
# Stopped at D:/Programming/github/vunit/examples/com/vunit_out/preprocessed/tb_
shuffler_lib/scoreboard.vhd 61 Process main
#
#
# Stack trace result from 'tb' command
#  D:/Programming/github/vunit/examples/com/vunit_out/preprocessed/tb_shuffler_l
ib/scoreboard.vhd 61 [address ff2c39c1] Process main
#
#
# Surrounding code from 'see' command
#   56 :     self := create("scoreboard");
#   57 :     subscribe(self, find("test runner"), status);
#   58 :     subscribe(self, find("monitor"), status);
#   59 :     loop
#   60 :       receive(net, self, message);
# ->61 :       case get_msg_type(message.payload.all) is
#   62 :         when reset_shuffler =>
#   63 :           n_received         := 0;
#   64 :           received_checksum  := (others => '0'); loaded_checksum := (ot
hers => '0');
#   65 :           received_checksum2 := (others => '0'); loaded_checksum2 := (o
thers => '0');
#
````

LarsAsplund, 2015-05-19:
@kraigher The wait statement in the wait_for_messages procedure in com.vhd is timing out immediately despite receive_timeout = 9223372036854775807 ps. The fatal error is because my tb doesn't check message.status and tries to decode a null payload. A negative timeout doesn't help us, we still need a very large timeout for the wait statement. I guess 1 hour would be large enough and it wouldn't overflow GHDL. Right?

```vhdl
procedure wait_for_messages (
  signal net               : in  network_t;
  constant receiver        : in  actor_t;
  variable status          : out com_status_t;
  constant receive_timeout : in  time := max_timeout_c) is
begin
  if messenger.deferred(receiver) then
    status := deferred_receiver_error;
  else
    status := ok;
    if not messenger.has_messages(receiver) then
      wait on net until messenger.has_messages(receiver) for receive_timeout;
      if not messenger.has_messages(receiver) then
        status := timeout;
      end if;
    end if;
  end if;
end procedure wait_for_messages;
```

kraigher, 2015-05-19:
@LarsAsplund The idea was not to simply pass the negative time onto the wait statement but rather to have an `if` statement surrounding the `wait` statement where if the timeout was negative the `for timeout` part would be removed from the `wait` statement. Anyway `1 hr` works for GHDL. That is what i have pushed to the branch.

kraigher, 2015-05-19:
Seems `(15 downto 4 => "Hello world!")` is a vhdl 2008 construct which GHDL does not support yet. 
https://sourceforge.net/p/ghdl-updates/tickets/72/

I will rewrite the code.

kraigher, 2015-05-19:
@LarsAsplund It seems time'low in GHDL is -2^63 instead of -2^63+1 as modelsim. You claim in a comment that the standard says it must be the symmetric range [-2^63+1, 2^63-1]. I cannot find this claim in the 2008 LRM.
> The only predefined physical type is type TIME. The range of TIME is implementation dependent, but it is
guaranteed to include the range –2147483647 to +2147483647.

Thus the encoding method does not work for `time'low` since the sign is flipped when it is negative.

LarsAsplund, 2015-05-20:
@kraigher Will your handling of the unsupported ranges affect the codecs ability to maintain the array range(s)?

My comment is badly phrased. What i meant was that Modelsim's position with a fixed integer range and a variable resolution is ok.

kraigher, 2015-05-20:
@LarsAsplund missing ticket 72 only affects the test bench: 22b0ef97dce866178e5c8e900bf40a70daeadea0

LarsAsplund, 2015-05-20:
@kraigher Ok, I don't mind those changes.

kraigher, 2015-05-20:
@LarsAsplund Found some more violations which caused GHDL errors and ModelSim warnings: e5810094777165a1ae6f8e8e343971ebc7b0935e
Using for example `bit'value("0")` is not valid it should be `bit'value("'0'")`.

kraigher, 2015-05-20:
@LarsAsplund There are failures with GHDL in `com_string.vhd` when data is range 3 downto -3:
```vhdl
  function to_detailed_string (
    constant data : ufixed)
    return string is
    variable unsigned_data : ieee.numeric_std.unsigned(data'length - 1 downto 0) := ieee.numeric_std.unsigned(data);
  begin
```

Modelsim also fails though not always:
```vhdl
    constant data : ufixed(3 downto -3) := to_ufixed(6.5, 3, -3);
    constant unsigned_data1 : ieee.numeric_std.unsigned(data'length - 1 downto 0) := ieee.numeric_std.unsigned(to_ufixed(6.5, 3, -3)); -- Works
    constant unsigned_data2 : ieee.numeric_std.unsigned(data'length - 1 downto 0) := ieee.numeric_std.unsigned(data); -- Fails
```

```
** Error: bug.vhd(16): (vcom-1167) Index value -3 (of type std.STANDARD.NATURAL) is out of range 0 to 2147483647.
```

kraigher, 2015-05-20:
I fixed the above with: fe2d854a4ecd0e7959c9f7950416399c2926ed81

kraigher, 2015-05-20:
Now the only failure in the com acceptance test is the encoding/decoding of `time'low`.

LarsAsplund, 2015-05-20:
Ok with the changes. Maybe a special case for time'low is the easiest solution? Or maybe the encoder can be built around the negative range rather than the positive range as it is today. Are you making those changes or do you want me to do it?

kraigher, 2015-05-20:
I will rewrite it.

kraigher, 2015-05-20:
@LarsAsplund I rewrote it to support non-symmetric time range. I still assumes time is 8 bytes though, one could compute a package constants based on time'high and time'low to decide how many bytes to use for time. I let I be since it works with both modelsim and ghdl . Here is the commi: 4e01cdddd03e8969362690c32cbca78a6683115a

kraigher, 2015-05-20:
Now everything works with GHDL except test_end_to_end.py were I will need to refator away the use of `vunit_finished`.

LarsAsplund, 2015-05-20:
@kraigher From the looks of it you made the effort to encode the 2's complement number (modulos of negative number always confuse me). If that's the case we should be able to handle 32 bit simulators (if they exist) and for those with wider ranges our test cases should fail so we wouldn't forget to fix it.

kraigher, 2015-05-20:
GHDL is now able to run all acceptance tests.  So we are very close at adding official support for it.

There is one last [bug]((https://sourceforge.net/p/ghdl-updates/tickets/77/) ) with std.env.stop that I have reported to them but which could be worked around. Hopefully they will fix it soon and the workaround can be removed. 

Other than that the only workaround left is the fixes in the OSVVM code. It is tricky what we should do about it. Adding those fixes on master might be sensitive without mentioning it in a note. I have reported the problems to Jim Lewis and he even sent me a preview of his next OSVVM release so that I could check it for GHDL errors. I sent him a report of the errors and how they could be fixed a couple of days ago but have not heard from him yet. We would also add ghdl to master without OSVVM fixes.

LarsAsplund, 2015-05-20:
Great work! I think we should avoid a modified version of OSVVM on master but I think it would be okay to have the GHDL support on master without supporting OSVVM yet. After all, it's an add-on. However, the best would be if we can have synchronized releases. Have you discussed when the next binary of GHDL will be released? Maybe we can await that?

Btw, have you built and tested GHDL+VUnit under Windows?

kraigher, 2015-05-20:
I have not asken when their next binary release arrives. It should be orthogonal to us having it on master. We say it works with GHDL built from latest development branch and which OS and backend we have tested with until we have an official release to test with or some other news to declare.

As for OSVVM Jim Lewis replied and said he has fixed the GHDL errors so hopefully there will be an official release of OSVVM that works with GHDL soon.

I have not tested GHDL on Windows yet since it is quite a project to build it from source there. I have the desire try some time in the near future.

kraigher, 2015-05-22:
I have managed to build GHDL mcode version on 32-bit Windows in VirtualBox using cygwin.
It runs everything except encode/decode of time with debug codecs. Seems there was an overflow in the decode function. I fixed it in b850c6121e5dced4346afe95b939478d53ae9a5d. The debug decode functions for time seems to be a workaround for modelsim not being able to do `time'value(time'image(time'low))` which works in GHDL by the way. With this fix everything runs with GHDL mcode.

Another minor problem I found with the mcode backend of GHDL was that it has no `-o` flag since no binary is written. Thus the GHDL simulator interface must detect which backend it is working against to omit the `-o` flag if it is mcode.

kraigher, 2015-05-22:
I added support for launching GTKWave and viewing the result after simulating with GHDL:
https://github.com/LarsAsplund/vunit/blob/ghdl/user_guide.md

[7 more replies]

closed 2015-06-01 by kraigher