VUnit/vunit #31: Memory leaks in string_ops.vhd

kraigher, 2015-03-02:
There are numerous occurrences in string_ops.vhd where the a 'write' is performed to a local 'line' variable. These operations allocate memory on the heap and the function must deallocate the 'line' before returning. 

In my project the test bench with the most check_equal:s crash now before completion now due memory allocation errors. Each check_equal allocates memory by calling the to_nibble_string which is never freed.

kraigher, 2015-03-02:
@LarsAsplund I assigned you since it is mostly your domain. Do you have the time to fix it or do you need help?

LarsAsplund, 2015-03-02:
@kraigher Yes, I've applied some lazy programming to avoid having to decide what a sufficiently long string is. I will set some max ranges so that you can avoid the crash. Maybe this should be configurable or we should have functions returning line rather than string?

kraigher, 2015-03-03:
@LarsAsplund Having a high max range is much better than having the memory leak. In many testbenches we do millions of check_equals. Also I think it should be possible to compute an exact or pessimistic string length in most cases (especially in to_nibble_string) completely removing the error-case:
```vhdl
function foo(a, b : integer) return string is
    variable str : string(1 to bar(a,b));
begin
    ...
    return str;
end function;
```

By the way another minor issue also came to mind when looking at this. We actually do create the failure string for every check_equal even if it does not fail. Otherwise I would not have discovered the memory leak since nothing failed. This could potentially slow down simulation a lot when you use it millions of times in a test bench such as checking each value on a wide data path every cycle. Can we not avoid this by only creating the the string in the conditional error path? A passing check only results in some statistics being incremented. Maybe adding a 'passing_check' function that has no error message parameter and represents an unconditionally passing check. This function could be called rather than the normal lower-lever check-function and the error message does not have to be created in that branch of the if-statement.

LarsAsplund, 2015-03-03:
@kraigher Totally agree that the leaks must be removed. The lazy style is only applicable in some closed projects. I'll scan the rest of the code for this. I know there are more places but I'm not sure if any of them have any practical implications today. I will also look at the suggested optimizations. All functions I fixed have upper bounds.

Could you test the current fix so that I can close this bug and handle the rest as enhancements?

kraigher, 2015-03-03:
@LarsAsplund I am running with the fix now while monitoring the memory consumption of the vsimk process and it does not increase over time as it did before. Thus the immediate problem is solved.

LarsAsplund, 2015-03-03:
@kraigher Good. I did the same analysis of a 10M iteration I ran.

closed 2015-03-03 by LarsAsplund