VUnit/vunit #32: Fault Tolerant Checkers

barri, 2015-03-09:
A neat feature when comparing different implementations of the same algorithm would be fault tolerant checkers. The checker would compare two values and allow for a user specified error tolerance. This would allow for easy comparison between different implementations vs some reference data or between each other. A bonus would be getting a report of minimum, maximum and average differences between the two value sets.

LarsAsplund, 2015-03-10:
The pure checker should be doable. It is a more generalised form of check_equal. However, I feel that collecting stats is something that should be done elsewhere.  Checks should be concerned with pass/fail. Stats would make it concerned with degrees of pass and fail. The question is how important this is to the user.  If that is important he/she would do the abs (measurement - value) calculations anyway and lot of the added convenience with a new checker would be lost. If you already have the diff you can do check_relation (diff < tolerance ). What do you think?

barri, 2015-03-10:
I think the pure checker would be useful to have for the cases where you don't need the stats. For the use cases where the stats are needed you can use the check_relation in the manner you suggested.

LarsAsplund, 2015-03-10:
Another comment. Even though a statistics module doesn't belong to the checker functionality it may still have a place in VUnit. Unit testing support is at focus of VUnit but we have not excluded other reusable components for testbench development. The communication mechanism is one such example. Given that testbenches based on a reference model is a common design pattern I would expect that there are reusable components in that area. Suggestions are appreciated.

LarsAsplund, 2015-03-10:
@barri  Right now I will focus on getting the communication mechanism ready. Please comment the latest additions!

However, let's start with some inputs for the "within tolerances checker"
- Would tolerances be symmetrical, i.e is a == b in ref-a <= measurement <= ref+b?
- What datatypes are you considering?
- I haven't given it any thought but would the addition of a tolerance parameter to check_equal be an acceptable solution? Default of that parameter would be zero. @kraigher Yor thoughts on this?

barri, 2015-03-10:
Supporting the signed and unsigned datatypes with symmetrical tolerances would probably cover the basic need. Adding a tolerance parameter to check_equal would be a a simple solution but it might be clearer to have a new function.

(Will try to catch up on the communication mechanism tonight.)

kraigher, 2015-03-10:
I would prefer not to add any extra parameters to check_equal as to not over-complicate those functions. I think this is something users could implement themselves with a small increment of functionality over the existing check-functions we provide and thus I am reluctant to add built-in support for it. It is better to leave it up to the user to decide what is "almost equal". If someone comes up with a nice package for this I will consider it but until then I am reluctant.

Another reason for my reluctance is that I personally not use this method of verification to test a synthesizable module since it could hide unwanted defects that creep into the code. A synthesizable module is typically obfuscated due to pipelining, instantiation of primtives and other implementation issues and I therefore want a bit-exact high level reference model of it. If two algorithms that are not exactly the same are to be compared they should be compared on a mathematical or software level and not at the RTL-level. The RTL should just be an exact replica of that high level algorithm where the numerical trade-offs have already been done.

LarsAsplund, 2015-03-11:
Agree with @kraigher that a bit exact model is the preferred situation but I would guess that many (most?) VHDL developers doesn't have such a model and, for one reason or the other, doesn't feel comfortable doing the refinement in the high-level language being used. We also have the case where there is no model at all, e.g. if you want to develop square root functionality and verify that by checking that sqrt(x) * sqrt(x) is almost equal x over a set of random values. So I think there are valid use cases. The question is more how much convenience such a check provides, i.e, the priority it would get in the todo list. To determine that I would like to investigate the use cases a bit more

@barri How much smartness would you like to see when it comes to identifying the position of the radix point? With plain unsigned/signed the check would have to assume that they are located in the same place. Have you considered packages like fixed_pkg? I'm not up-to-date with how well that is supported in various synthesis tools

We have talked about having a tolerance as input but there are also other ways. This is the PyUnit for Python way. Preferences?

```console
assertAlmostEqual(first, second, places=7, msg=None, delta=None) 
assertNotAlmostEqual(first, second, places=7, msg=None, delta=None) 
Test that first and second are approximately (or not approximately) equal by computing the difference, rounding to the given number of decimal places (default 7), and comparing to zero. Note that these methods round the values to the given number of decimal places (i.e. like the round() function) and not significant digits.

If delta is supplied instead of places then the difference between first and second must be less or equal to (or greater than) delta.

Supplying both delta and places raises a TypeError.
```

barri, 2015-05-07:
I apologize for the late answer.
I don't expect much smartness at all when it comes to the checker. The assertAlmostEqual with just a delta was what I was looking for when I wrote this issue. Supporting fixed point would be a cool feature but not really a necessary one.

LarsAsplund, 2015-06-11:
@barri I started to look at this but think I'm about to change my mind. With `check_equal` we support comparison between different types which results in 70 different checks to support "all" combinations. Now that we want a third delta input parameter the number of combinations explodes. We also need two subprograms for each combination (check_almost_equal and check_not_almost_equal). I could limit myself to only support subprograms where all three inputs have the same type but that would limit the useability. The added value of the remaining subprograms would also be lowered because now it becomes easy for the user to use predefined functions and operators to create a custom check based on the expression `abs(expected - got) <= delta`. Comments on this?

barri, 2015-06-17:
Valid points, might as well close this issue. Thanks for your time
considering my suggestion.

kraigher, 2016-03-18:
I think we should add a `check_almost_equal(got, expected, eps : real)` to check that `abs (got - expected) < eps` such as:
```vhdl
procedure check_almost_equal(got, expected, eps : real) is
  constant difference : real := abs (got - expected);
begin
  assert difference < eps
    report ("abs(got - expected) was (" & to_string(got) & " - " & to_string(expected) & ") = " &
          to_string(difference) & " not less than eps = " & to_string(eps)) severity failure;
end procedure;
```
This is useful when testing modules where the result is correct if it is close enough to some floating point model. For instance a module that should compute arctan with an epsilon of 2**-9 compared to a standard floating point arctan such as the one from ieee.math_real or MATLAB.

We should only support real types. The user has to convert his std_logic_vector or whatever to a real anyway to divide with the implicit fixed point gain.

cmarqu, 2016-03-18:
FWIW, some such functions allow `eps` to be defined in terms of decimal digits which seems useful.
Other ways to specify `eps` I have seen or implemented are percentage of `expected` or even percentage of full scale (which would be another parameter in the function signature).

LarsAsplund, 2016-03-19:
@kraigher Any preferences when it comes to < or <=? Equality is a bit fuzzy with floats so maybe it's just a matter of what feels more natural. @cmarqu Yes, pyunit use delta or decimal places. Default is to use places but delta comes more natural to me. Anyway, if we're going to support both I would prefer different procedures/signatures.

kraigher, 2016-03-19:
I think "correct decimal places" or "correct number of bits" is useless and an ill-defined if it does not mean eps = 10^-n or eps = 2^-n in the formula I wrote above. If we agree on this then there is no need to have different functions for decimal places. The user can write eps=10^-n or 2^-n themselves if they want to think in decimal or binary precision. 

Regarding equality. Most sources only seen to use strictly less than rather than less than or equal and I prefer it as well. When thinking in significant binary digits and one of the operands is a actually a fixed point number it makes sense since if the difference was exactly 2**-n then it should be the next or previous fixed point number.

Regarding relative vs absolute error there seem to be two types of formulas:

1. abs(x - y) < eps
2. abs(x - y) < eps * max(abs(x), abs(y))

The second formula is generally recommended when you are actually using real floating point numbers in the DUT as it takes the huge dynamic range of floating point values into account. However in the specific domain of DSP and fixed point calculations the first formula is better as either x or y is really a fixed point number with a known scale with a requirement that it should be accurate to n bits which means eps = 2^-n.

So we should have both formulas. The first one is recommended for when the DUT computes fixed point values and the other when you actually have floating point operations in the DUT.

LarsAsplund, 2016-03-19:
In Python < is used and `assertAlmostEqual` is the same as almost equal to n digits means that the abs difference is < 0.5*10^-n.
```
>>> a.assertAlmostEqual(1, 1.05,places=1)
Traceback (most recent call last):
  File "<pyshell#21>", line 1, in <module>
    a.assertAlmostEqual(1, 1.05,places=1)
  File "C:\Python34\lib\unittest\case.py", line 842, in assertAlmostEqual
    raise self.failureException(msg)
AssertionError: 1 != 1.05 within 1 places

>>> a.assertAlmostEqual(1, 1.05,delta=0.05)
Traceback (most recent call last):
  File "<pyshell#23>", line 1, in <module>
    a.assertAlmostEqual(1, 1.05,delta=0.05)
  File "C:\Python34\lib\unittest\case.py", line 842, in assertAlmostEqual
    raise self.failureException(msg)
AssertionError: 1 != 1.05 within 0.05 delta
```

kraigher, 2018-02-23:
I added a `check_equal` for reals now that compares the absolute value with a `maxdiff`. The `maxdiff` has a default value of 0.0.

closed 2018-02-23 by kraigher