VUnit/vunit #33: Incompatible packages?

philippefaes, 2015-03-19:
Hi, I'm working to set up the logging library; I will proceed to other parts later. I notice that several packages are declared in more than one file, and have incompatible declarations.

For example, some package declarations for "log_base_pkg" have the type "log_call_args_t", while others do not. Now some tests (tb_dictionary and test_support) use this type, which breaks the compilation.

Is it intentional that package have incompatible API? What is the convention? Which package should I include in my projects?

LarsAsplund, 2015-03-19:
@philippefaes 
The reason for having several declarations is partly to support all VHDL versions from 93 to 2008 and partly for the testing of VUnit. For some test suites we use mocks, e.g. when unit testing check we don't use the normal log package but a mock that will catch and verify the log calls we expect from the check package. When using the VUnit ui to compile it will only add the files representing the normal functionality for the VHDL version you have specified. If you want to compile separately you can have a look how it's done in ui.py:

```python
        if mock_lang:
            files += [join("vhdl", "src", "lang", "lang_mock.vhd")]
        else:
            files += [join("vhdl", "src", "lang", "lang.vhd")]

        files += [join("vhdl", "src", "lib", "std", "textio.vhd"),
                  join("string_ops", "src", "string_ops.vhd"),
                  join("check", "src", "check.vhd"),
                  join("check", "src", "check_api.vhd"),
                  join("check", "src", "check_base_api.vhd"),
                  join("check", "src", "check_types.vhd"),
                  join("run", "src", "run.vhd"),
                  join("run", "src", "run_api.vhd"),
                  join("run", "src", "run_types.vhd"),
                  join("run", "src", "run_base_api.vhd")]

        files +=  [join("logging", "src", "log_api.vhd"),
                   join("logging", "src", "log_formatting.vhd"),
                   join("logging", "src", "log.vhd"),
                   join("logging", "src", "log_types.vhd")]

        files +=  [join("dictionary", "src", "dictionary.vhd")]

        files +=  [join("path", "src", "path.vhd")]

        if self._vhdl_standard == '93':
            if mock_log:
                files += [join("logging", "src", "log_base93_mock.vhd"),
                          join("logging", "src", "log_special_types93.vhd"),
                          join("logging", "src", "log_base_api_mock.vhd")]
            else:
                files += [join("logging", "src", "log_base93.vhd"),
                          join("logging", "src", "log_special_types93.vhd"),
                          join("logging", "src", "log_base_api.vhd")]

            files += [join("check", "src", "check_base93.vhd"),
                      join("check", "src", "check_special_types93.vhd"),
                      join("run", "src", "run_base93.vhd"),
                      join("run", "src", "run_special_types93.vhd")]

        elif self._vhdl_standard in ('2002', '2008'):
            if mock_log:
                files += [join("logging", "src", "log_base.vhd"),
                          join("logging", "src", "log_special_types200x_mock.vhd"),
                          join("logging", "src", "log_base_api.vhd")]
            else:
                files += [join("logging", "src", "log_base.vhd"),
                          join("logging", "src", "log_special_types200x.vhd"),
                          join("logging", "src", "log_base_api.vhd")]

            files += [join("check", "src", "check_base.vhd"),
                      join("check", "src", "check_special_types200x.vhd"),
                      join("run", "src", "run_base.vhd"),
                      join("run", "src", "run_special_types200x.vhd")]

            if self._vhdl_standard == '2008':
                files += ["vunit_context.vhd"]
```

Under a normal compilation mock_lang and mock_log will be false so all files with mock in the file name are ignored. Then there are a number of files common to all VHDL versions and some files that are specific to the version you're using.

Did this help you?

philippefaes, 2015-03-20:
Thanks Lars. I got the library to compile now. I'm choosing the 2008-compatible files, without mock. The only thing that doesn't work yet is the "test" files inside the vunit_lib (in the "vhdl/*/test" directories). It seems that these files can only be compiled against the "mock" versions of the library. Is that correct?

LarsAsplund, 2015-03-20:
That's correct @philippefaes . If you look at the acceptance test for the check package (test_check.py) you will see that that python script has a special line

```python
ui.add_builtins('vunit_lib', mock_log=True)
```

That enables the log mock during that test. What simulator are you using? If using Modelsim I encourage you to use the scripting capabilities  in VUnit which makes all this much easier. Aldec and GHDL support is around the corner.

LarsAsplund, 2015-06-11:
The questions have been answered and the discussions have ended so I'll put this in the archive.

closed 2015-06-11 by LarsAsplund