VUnit/vunit #36: Report ModelSim bugs to Mentor Graphics

kraigher, 2015-04-01:
# Bug 1
There is a bug in ModelSim setting the MODELSIM environment variable directly within TCL before executing a vsim command. The bug has been found in at least Microsemi Modelsim 10.3a and Altera Modelsim 10.3c (Altera 14.1 toolchain). 

Example of failure:
```tcl
set env(MODELSIM) path/to/modelsim.ini
vsim lib.entity # fails since "path/to/modelsim.ini" is not read
echo $env(MODELSIM) # also fails and says there is no MODELSIM environment variable
```
Interestingly it fails even if the MODELSIM environment variable is correctly set by the parent process of ModelSim. So the problem is not just that the set env(MODELSIM) command does not take effect, it actually seems to destroy something within the program. 

Example of work-arround:
```tcl
set env(MODELSIM) path/to/modelsim.ini
printenv; echo $env(MODELSIM) # Add this between 'set' and 'vsim' and it works... 
vsim lib.entity # Works
```

# Bug 2
The -modelsimini flag does not support a path with spaces even if it is correctly escaped. A similiarily escaped path works with other flags such as -wlf or -t.

# Our problems because of these bugs
To be able to run the generated tcl files manually in ModelSim we want to have the setting of the MODELSIM environment variable in the TCL file. We have to use the MODELSIM environment variable since the -modelsimini flag does not work with paths containing spaces. Removing the setting of MODELSIM environment variable within the TCL script will solve the problem for the affected ModelSim versions when running through Python but the TCL files will not work stand alone.

kraigher, 2015-04-01:
Getting in contact with Mentor Graphics and report this seems harder than getting in contact with some letterbox company front but I will give it a try through my work related channels.

kraigher, 2015-04-01:
Removed setting of MODELSIM environment variable directly in TCL in commit 80fc4b5e8a9bb0e90b73c132cf2e5b7fb6eac8d7. The risk is just to high that this will cause problems for VUnit users and hurt the adoption. This effectively removes support for using the user.do TCL script stand-alone directly from modelsim but this is not documented or officially supported anyway. Using the --gui flag which is the officially documented method will work since MODELSIM environment variable is set in the python parent process.

nathanaelhuffman, 2015-04-02:
Can you post a -modelsimini example of the error for bug2 as well?  I glanced around Mentor's supportnet site and didn't find examples of these.  I'm on 10.1a currently but I can install a newer version of modelsim tomorrow and if I can recreate will reach out to my FAE or create a service request.

kraigher, 2015-04-02:
@nathanaelhuffman I added some more details on the -modelsimini flag bug

nathanaelhuffman, 2015-04-02:
@kraigher For bug1 above, I can't reproduce in windows in ModelSim 10.4a SE. I ran the user_guide lib.tb_example by doing the following (after having used run.py 1x):
```bash
cd vunit\examples\user_guide\vunit_out
vsim.exe -do ./tests/lib.tb_example/msim/batch.do
```
And it properly launches modelsim and I see the Hellow world! in the transcript

For bug2 above, I can reproduce in .  What I did notice is that using double quotes does seem to work though without any escaping:
```bash
vsim -modelsimini "path with space/modelsim.ini"  # Works ok
```

I can still reach out to my mentor guy about the inconsistency in -modelsimini but if double-quoting works it may be a nicer workaround.

kraigher, 2015-04-02:
@nathanaelhuffman As I mentioned above double quotes also cause the same problem with the -modelsimini flag using Altera Starter Edition 10.3c so it is not a valid work-arround.

nathanaelhuffman, 2015-04-02:
@kraigher missed the comment about double quotes in the block above.  I have a better test case (user error last time) and bug2 can be reproduced in 10.4a and double quotes does not help.

Bug1 appears to be fixed in 10.4a but I did reproduce in 10.3c after installing that.  Reached out to my FAE, we'll see what he says.

kraigher, 2015-04-02:
@nathanaelhuffman Great, thank you!

nathanaelhuffman, 2015-04-03:
@kraigher Mentor support will be filing a bug for bug#2.  Apparently there is one way to make the -modelsimini flag work with spaces:
```bash
vsim –modelsimini “path/with a/space/modelsim.ini” –do my.do
```
but vsim can't deal with nested -do arguments so my.do would need to have all of the simulator commands in it, and I'm deep enough into vunit to decide whether that's worth looking into or not, cursory look says that would affect the way the do files are structured, and it's unclear to me if that would break the non-interactive session.

Other ideas were putting the ini file in the simulation directory and not using (unsetting) the modelsim env variable or the -modelsimini flag or putting the ini file in a place where a relative path without spaces can be used.

kraigher, 2015-04-04:
@nathanaelhuffman I think we can live with the current workarround since de .do files do not need to be stand-alone as of adding the --gui flag. When running from Python the environment variable is set correctly before launching the vsim process. It is good that they know about the bug though so that it can be an option to use -modelsimini flag in the future.

kraigher, 2015-05-30:
@nathanaelhuffman Have you got any more feedback from Mentor regarding these bugs such as if they were fixed or not?

nathanaelhuffman, 2015-06-03:
@kraigher No update. I can see the issue in Mentor's supportnet site but it's still "Not scheduled" for a planned release.  The defect number is DVT-76776.
Here is the list of possible workarounds provided from Mentor, you can see if one of these can be shoe-horned to work in vunit:
1.       Setting the MODELSIM environment variable in the OS before launching ModelSim. For some reason this inherits the env variable correctly.
2.       Launching ModelSim by specifying the -modelsimini "path/with/a space/modelsim.ini", but *not* launching a design. E.g. vsim -modelsimini "path/with/a space/modelsim.ini"
3.       Launching ModelSim with -modelsimini, but adding in the simulation directive with a -do call. E.g. vsim -modelsimini "path/with/a space/modelsim.ini" -do "vsim mydesign"

kraigher, 2015-06-03:
@nathanaelhuffman We use 1. which works fine. With the bugs reported such that mentor can improve their tool this issue can now be closed. Thank you for your help.

closed 2015-06-03 by kraigher