VUnit/vunit #39: database can not be created

KevinKes, 2015-04-25:
I just downloaded the master tree, but when I ran the examples I got the following error. Tried on different locations, as well with a different user, but no effect.
```
C:\vunit-master\examples\user_guide>python run.py
Traceback (most recent call last):
  File "run.py", line 18, in <module>
    ui = VUnit.from_argv()
  File "..\..\vunit\ui.py", line 96, in from_argv
    simulator_name=args.sim)
  File "..\..\vunit\ui.py", line 220, in __init__
    self._create_project()
  File "..\..\vunit\ui.py", line 229, in _create_project
    database = self._create_database()
  File "..\..\vunit\ui.py", line 252, in _create_database
    database = shelve.open(project_database_file_name, flag='n')
  File "c:\Python27\lib\shelve.py", line 239, in open
    return DbfilenameShelf(filename, flag, protocol, writeback)
  File "c:\Python27\lib\shelve.py", line 223, in __init__
    Shelf.__init__(self, anydbm.open(filename, flag), protocol, writeback)
  File "c:\Python27\lib\anydbm.py", line 85, in open
    return mod.open(file, flag, mode)
  File "c:\Python27\lib\dbhash.py", line 18, in open
    return bsddb.hashopen(file, flag, mode)
  File "c:\Python27\lib\bsddb\__init__.py", line 364, in hashopen
    d.open(file, db.DB_HASH, flags, mode)
bsddb.db.DBPermissionsError: (1, 'Operation not permitted')
```

kraigher, 2015-04-25:
@KevinKes Could you try the database_issue branch and see if it fixes the problem. It could be that a database is still open when a re-creation is attempted.

KevinKes, 2015-04-26:
@kraigher The database_issue branch fixes the problem for me.

commit 854d46e0 (2015-04-26): Closing database before re-opening. Closes #39.
closed 2015-04-26 by kraigher