VUnit/vunit #40: recursive logging of aall signals in GUI-mode

gunnarstjernberg, 2015-04-26:
When invoking the simulation in GUI, it would be a nice feature to be able to submit an argument that is equivalent to the ModelSim/QuestaSim command "log -r /*"

That way I could run the simulation and on ly afterwards bring up the signals I might be interested in for visual debugging.

By making a VUnit argument I need not care what the syntax is in different simulators which makes it kind of useful. However, I'm not sure iSim and/or VivadoSim supports an equivalent command.

kraigher, 2015-04-26:
@gunnarstjernberg It is already possible to do this in the Modelsim GUI mode which is currently the only supported simulator:

Just do:
```tcl
vunit_load
log -r /*
vunit_run
```

Instead of:
```tcl
vunit_load
vunit_run
```

The intention is to be able add logging of signals after the vunit_load command which is what I do all the time at work. To re-run the test there is no need to do vunit_load again just do:
```
restart -f;
vunit_run
```

Do you really think it is worth having this as a command line flag to VUnit when it can be achieved using the method above? It just saves you one line of typing or one right-click in the GUI. The point of the GUI mode is to let the user control how he wants to debug the test case and I would be a bit reluctant towards adding many specific command line flags to VUnit to replace one-liners in the GUI TCL window even if your request seems like a common user desire.

kraigher, 2015-04-26:
@gunnarstjernberg I added a section in the user guide about running test cases in the GUI:
https://github.com/LarsAsplund/vunit/blob/master/user_guide.md#running-a-test-case-in-the-modelsim-gui

kraigher, 2015-04-26:
@gunnarstjernberg Maybe a --gui-run flag that automatically does vunit_load, adds all signals to the log, did vunit_run and then stopped for inspection would be a time saver in many cases.

gunnarstjernberg, 2015-04-27:
Well, its no big issue. But having a simulator-independent (when more simulators are supported) way of doing it would save some time and increase easy-of-use imo.

Definitely not a requirement, but a "nice-to-have" feature.Just my 2 cents... :-)

kraigher, 2015-04-27:
@gunnarstjernberg What do you think about the --gui-run flag that I proposed above? That way there are two flags --gui and --gui-run where the first would just open the gui with the right configuration without running anything as it does today and the second would open the gui and automatically run the test after first adding all signals to the log.

gunnarstjernberg, 2015-04-27:
That would offer the capabilities that I'm requesting so yes, that is a good idea.

kraigher, 2015-04-27:
@gunnarstjernberg I will add that flag then.

kraigher, 2015-04-27:
@gunnarstjernberg I have added the functionality in 78df5ee9621c31090ce6f7c9fd94aa096de366b0:
https://github.com/LarsAsplund/vunit/blob/master/user_guide.md#running-a-test-case-in-the-modelsim-gui

Now you can do `--gui=load` which only launches the gui and loads the design awaiting manual configuration and run trigger by the user or `--gui=run` which automatically runs the test case logging all signals and variables recursively.

closed 2015-04-27 by kraigher