VUnit/vunit #41: VUnit crashes on 'é' in source file comments with Python 2.7

KevinKes, 2015-04-30:
using Python 2.7 VUnit crashes if the character 'é' is found in the comments of the VHDL file.
When Python 3.4 is used, everything works fine.

For example when I run the user_guide example with a modified tb_example.vhd to include the char é on line 4, I get the following error:
```
Traceback (most recent call last):
  File "C:\Users\kevin\Documents\GitHub\vunit\examples\user_guide\run.py", line 20, in <module>
    lib.add_source_files(join(root, "*.vhd"))
  File "C:\Users\kevin\Documents\GitHub\vunit\vunit\ui.py", line 549, in add_source_files
    self._parent.add_source_files(pattern, self._library_name, preprocessors)
  File "C:\Users\kevin\Documents\GitHub\vunit\vunit\ui.py", line 320, in add_source_files
    file_type=file_type_of(file_name))
  File "C:\Users\kevin\Documents\GitHub\vunit\vunit\project.py", line 73, in add_source_file
    source_file = SourceFile(file_name, library, file_type, vhdl_parser=self._vhdl_parser)
  File "C:\Users\kevin\Documents\GitHub\vunit\vunit\project.py", line 349, in __init__
    self._content_hash = hash_bytes(code.encode())
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 205: ordinal not in range(128)
```

kraigher, 2015-04-30:
@KevinKes One of the major differences between Python 2.x and Python 3.x is the handling of strings, unicode and bytes. It is one of the pain points of maintaining a Python application that has to support both. Anyway I added a acceptance test for this and added a special case for Python 2.x so I believe it is fixed now. I pushed this to the `issue_41` branch. Could you please verify that this fixes your issue after which I will merge it to master branch.

KevinKes, 2015-05-01:
@kraigher This fixes my problems. Thanks! Keep up the good work.

kraigher, 2015-05-01:
@KevinKes I pushed the fix to the master branch.

closed 2015-05-01 by kraigher

suoto, 2016-08-10:
This issue also happens on Python 2.7 when the location preprocessor writes a file with non-ASCII characters.

The [`six`](https://pythonhosted.org/six/) package provides some compatibility helpers for Python 2 and 3, maybe it could be used to handle some string conversions on VUnit as well.

commit 70909d5e (2015-05-01): Fixed exception on non-ASCII encoded files using Python 2.7. Closes #41.