VUnit/vunit #43: Ignored Python Runner Exception

barri, 2015-05-07:
When using the python runner to run test cases this ignored exception will sometimes appear:

Exception ignored in: <bound method DbfilenameShelf.__del__ of <shelve.DbfilenameShelf object at 0x00AD9830>>
Traceback (most recent call last):
  File "c:\Python34\lib\shelve.py", line 158, in __del__
  File "c:\Python34\lib\shelve.py", line 141, in close
  File "c:\Python34\lib\shelve.py", line 168, in sync
  File "c:\Python34\lib\dbm\dumb.py", line 103, in _commit
TypeError: 'NoneType' object is not callable
Exception ignored in: <bound method _Database.close of <dbm.dumb._Database object at 0x02E9E810>>
Traceback (most recent call last):
  File "c:\Python34\lib\dbm\dumb.py", line 250, in close
  File "c:\Python34\lib\dbm\dumb.py", line 103, in _commit
TypeError: 'NoneType' object is not callable

This trace was generated using the newest commit on the master branch but has been appearing for a few weeks. There was no vunit_out folder generated when the script was ran but it can appear even though there already is a vunit_out folder generated. The number of test cases ran seem to have no impact on the exception, I have gotten the error with 0 test cases ran as well.

kraigher, 2015-05-08:
@barri We are using the `shelve` Python library for the persistent database of VHDL parse results. This is to speed up run.py since VHDL parsing in Python is slow. On Windows the database backend used for `shelve` seems a bit fishy, it is even called dumbdbm. From dumbdbm code documentation:

   **Note** The dumbdbm module is intended as a last resort fallback for the anydbm module when no more robust module is available. The dumbdbm module is not written for speed and is not nearly as heavily used as the other database modules.

Thus I have written my own simple file based database to replace `shelve` and pushed it to the `new_database` branch. Could you try it out?

kraigher, 2015-05-08:
@barri And do not forget to do `--clean` the first time.

barri, 2015-05-08:
Seems to work fine.

commit 629dc929 (2015-05-08): Added a new database implementation to replace flaky dumbdbm. Closes #43.
closed 2015-05-08 by kraigher