VUnit/vunit #44: Add NVC simulator support

kraigher, 2015-05-09:
I am in progress of adding support for the [NVC simulator](https://github.com/nickg/nvc). There are still some bugs and features to be fixed be before we can add support for it but the author is very cooperative and is making great progress.

kraigher, 2015-05-12:
I have pushed the work in progress to the `nvc` branch. The currently blocking bug is https://github.com/nickg/nvc/issues/185 which is triggered by `test_runner_cleanup`. Otherwise it is able to compile and run the user_guide example up until `test_runner_cleanup`.

peteut, 2016-04-02:
I just grepped the `nvc` branch but couldn't find any reference to `nvc`. Is there any progress on this?

kraigher, 2016-04-02:
@peteut `nvc` support has not been quite finished yet but the basic user guide example works. There are still some problems to be solved like that it does not like some of our custom resolved types in the VHDL runner but @LarsAsplund might refactor those away. It is in a usable state though unless you use special stuff like `test_runner_watchdog` which will require the custom resolved type.

peteut, 2016-04-02:
Great, I'm testing my `VHDL` backend for [Migen](https://github.com/peteut/migen/tree/vhdl) currently and aiming for unit test support, `vunit` looks like the perfect fit. I'm not going to use `test_runner_watchdog` anyway and wrote a basic `SimulatorInterface` for `nvc` for now.

kraigher, 2016-04-02:
@peteut There is already a simulator interface for nvc on the nvc branch.

peteut, 2016-04-02:
@kraigher: got it, thanks! I somehow created a local branch, instead of checking out your `nvc` branch. Sorry for the confusion.

nickg, 2018-02-05:
@kraigher what issues are stopping this working now? I have some free time to work on nvc at the moment so could look into any problems you run into...

kraigher, 2018-02-06:
@nickg I will check with your latest fixes after I rebase the NVC branch. At least this issue is still a blocker https://github.com/nickg/nvc/issues/202 if we do not work arround it in VUnit itself.

nickg, 2018-02-06:
@kraigher nickg/nvc#202 is fixed now but I only tested it it a few simple cases so it may still have bugs

kraigher, 2018-02-06:
@nickg I rebased nvc branch on vunit3 and found various problems. I will create reprocers for them. Some assets failed and some unresolved symbol.

LarsAsplund, 2023-02-27:
Closed by #904

closed 2023-02-27 by LarsAsplund