VUnit/vunit #48: Parallel test execution

kraigher, 2015-06-03:
Support parallel test execution with a `-p <num>` flag. Continuous output such as with the `-v` flag would not be possible when running in parallel.

kraigher, 2015-06-08:
Parallel test execution works well now on the `parallel` branch. I ran into a lot of Windows specific issues reminding me why it is such a crappy operating system from a programmers perspective. There seems to be some deadlock left to debug when pressing Ctrl-C on Windows but it is very rare. 

I also accidentally found out that using Git Bash (MSYS Bash) on windows will [hard kill the process when pressing Ctrl-C](http://stackoverflow.com/questions/23678045/control-c-kills-ipython-in-git-bash-on-windows-7) after a very short time prohibiting VUnit from making a graceful shutdown and terminate its vsim child processes potentially leaving them open after Ctrl-C. There is a work-around that can be put in .bashrc but we cannot solve it from within VUnit, cmd.exe works fine with Ctrl-C though.

I will continue to explore the rare Ctrl-C deadlock on Windows for a while before I merge this. But at this point I would not advice anyone against using this branch if they want to start running tests in parallel.

kraigher, 2015-06-08:
There is also a race condition with ghdl when running multiple simulations in parallel with the same top level due to temporary files it creates. https://sourceforge.net/p/ghdl-updates/tickets/87/

LarsAsplund, 2015-06-08:
Agree on merging this despite the problems you mention. This is a very nice
feature for anyone not affected by the exceptions.

kraigher, 2015-06-10:
The GHDL issue has now been fixed so that one can use parallel execution without race conditions.

closed 2015-06-10 by kraigher