VUnit/vunit #50: Incomplete dependency scanning for configuration

imd1, 2015-06-15:
Using V0.20.0

When a configuration has a "use entity lib.ent(arch)" statement, the file containing the configuration currently correctly depends on the file containing ent.  However, the configuration may continue with "for arch" statements containing "for all:arch_comp" statements which requires that the configuration file to also depend on the file contain arch, otherwise vcom errors about arch_comp not being recognised as components.  This does not appear to work currently, so there is a race between the arch file compilation and the configuration file compilation.

kraigher, 2015-06-15:
I am pretty sure that "use entity lib.ent(arch)" will insert dependency between both the file containing ent and the file containing arch. Could you provide a small VHDL example to pinpoint exactly the problem?

imd1, 2015-06-15:
I'm away from the code, so I'll try and write some (pseudo) code for now:

File A.vhd:
configuration cfgA of A
   for all: bar use entity lib.B(archB)
      for archB
          for all: archB_comp use ....
      end for;
end configuration;

File B.vhd:
entity B
end entity;

File C.vhd:
architecture archB of B
component archB_comp;
begin
foo: archB_comp;
end architecture;

A.vhd correctly depends on B.vhd but does not and should also depend on C.vhd. When vcom processes A.vhd, I get an error message about archB_comp not recognised as a component when C.vhd has not previously been compiled.

Let me know if this clarifies the issue or not.

kraigher, 2015-06-16:
I tested your exact example (minus some syntax errors) and `A.vhd` depends on both `B.vhd` and `C.vhd` as I already assumed it would based on how I know this is implemented.

I added the following test to verify and it passed:
```python
    def test_issue_50(self):
        self.project.add_library("lib", "lib_path")

        self.add_source_file("lib", "A.vhd", """
configuration cfgA of A
for all: bar use entity lib.B(archB)
for archB
for all: archB_comp use ....
end for;
end configuration;
""")

        self.add_source_file("lib", "B.vhd", """
entity B is
end entity;
""")

        self.add_source_file("lib", "C.vhd", """
architecture archB of B is
component archB_comp;
begin
foo: archB_comp;
end architecture;
""")
        self.assert_compiles("C.vhd", before="A.vhd")
        self.assert_compiles("B.vhd", before="A.vhd")
```

There must be something more to it than what you have posted.

imd1, 2015-06-16:
Thanks very much - I will look into this further when I have access to the code later

imd1, 2015-06-16:
I've mocked up the example I gave you and agree that the scanning is OK for this example.  I've also studied the real example and still think there's an issue.  The real example is more complicated, so I will try and create a mock that exhibits the issue in the same way - this may take some time as this is very much a part time activity - so please bear with me.

imd1, 2015-06-17:
I have taken the real code, pruned it to a reasonable number and complexity of files, and maintained the issue - please let me know how I can send you the ZIP file so you can work on this further

kraigher, 2015-06-17:
You can email it to me. My email is listed in my github profile.

kraigher, 2015-06-17:
@imd1 I found the problem. The dependency scanner regexp did not handle a space between the entity and architecture names. `work.Register_fifo (Rtl)` did not work but `work.Register_fifo(Rtl)` worked. I will fix it and make a new release.

kraigher, 2015-06-17:
In release v0.23.0
https://github.com/LarsAsplund/vunit/releases/tag/v0.23.0

closed 2015-06-17 by kraigher