VUnit/vunit #51: Aiutomaticlly execute each test in parallel

imd1, 2015-06-20:
The recently added parallel test execution feature is great - thanks - and I tend to set the number equal to the number of tests.  Is there a way to do this automatically so the number changes as tests are added?

imd1, 2015-06-20:
Apologies for the heading error!

LarsAsplund, 2015-06-20:
Let's look into that. Just out of curiosity, what performance gain do you see? With what simulator, OS, hardware, and so on?

kraigher, 2015-06-20:
You can change the heading by pressing the edit button to the right of the issue heading.

Regarding the subject matter. Are you sure it makes sense to set the number of threads equal to the number of tests? Setting it to the number of processor cores or slightly higher is probably the best performance option and that is regardless of how many tests you run.

imd1, 2015-06-20:
I'm running 8 tests on a Jenkins server.

My knowledge of its capabilities is limited at this moment. It runs Win7 and has 32GB of memory. I can find out more this week.

So far, I am seeing a linear benefit in setting the number of threads equal to the number of tests. I am happy to continue manually setting this parameter. I am no expert in this field. Is it not possible to achieve performance benefit by multi threading on a single core?

LarsAsplund, 2015-06-20:
It's possible but there's a limit, see https://groups.google.com/forum/?nomobile=true#!topic/comp.lang.vhdl/mJhz5kBvZlA so i agree with Olof on this

closed 2015-06-20 by imd1

kraigher, 2015-06-20:
Be careful when looking at the total vs elapsed time printed by VUnit to make any speedup conclusions. The ratio with high P will always approach P but when the actual speedup stops it will be due to the total time increasing where as the elapsed time stays the same. The number optimal number of cores will really only depend on the machine rather than the number of tests.