VUnit/vunit #53: Positive Acknowledge of Each Check

LarsAsplund, 2015-06-30:
Some test libraries like BVUL and OSVVM provide a way to have a postive acknowledge on each passed check and not just a negative acknowledge when it fails. Is this something you would like to see in VUnit as well or is positive acknowledge of the test cases sufficient level of details? Provide your use cases for why we should have it or the reasons for why we shouldn't. See also https://groups.google.com/forum/#!topic/comp.lang.vhdl/t7BmFSUiNpg.

LarsAsplund, 2015-09-08:
No online comments so far and the only offline use case I've heard is the ability to automatically check which requirements that have been tested and whether or not they passed or failed. This can be done if the check message is tagged with the requirement ID and then you do parsing of the test report. However, with test cases verifying a well-defined behavior there's a better mapping between the test case name and a requirement than it is between the error message of a check and a requirement. For example, this is a test case from the `com` package testbench

``` vhdl
elsif run("Test that two actors of the same name cannot be created") then
  actor := create("actor2");
  check(actor /= null_actor_c, "Failed to create named actor");
  check(create("actor2") = null_actor_c, "Was allowed to create an actor duplicate");
```

A report with

``` console
Pass Test that two actors of the same name cannot be created
```

is very readable while making a pass statement from an error message may be less so

``` console
Pass Failed to create named actor
``` 

Since we have a pass/fail report on the test case level it's probably a better strategy to tag the test case name with a requirement ID.

Another problem with the tagging approach regardless if it's applied to test cases or check error messages is that they may not have a one-to-one mapping with a requirement. From what I've seen people tend to place the requirements as tags in a comment next to the test case rather than having multiple requirement IDs in the name itself. Parsing is than done of the source code and the test report to get the complete picture.

Currently, with no significant interest and a weak use case there is no motivation for adding such a functionality.

cmarqu, 2016-01-29:
I could see it useful to have this functionality when a colleague has a testcase that eventually fails at some point, then I can ask for the full log and possibly see from the context why the failure occurs, without having to repeat the simulation. But that certainly depends on the rest of the logging output too.

LarsAsplund, 2016-01-31:
@cmarqu With that use case I'm thinking that a passing check should be logged as a debug message and that you would configure debug messages to be passed to file while being suppressed on stdout. The example above which just put `Pass` before the error message was not very readable so I would probably want something more verbose. Assuming I enable location preprocessing and set the output format to `verbose` a passing check like this

``` vhdl
check_equal(data, 32);
```

could result in a log entry like this

```
1000 ps: DEBUG in (some_file.vhd:147): Passed check call with message "Equality check failed! Got 0010_0000 (32). Expected 32 (0010_0000).".
```

Logs on passing checks would obviously be something that can be configured and the default would be not to log. What more should be configurable?
- The log level? I used debug but should it be possible to set some other level, possibly a custom level, to more easily distinguish these entries from others?
- The extra "Passed check call with message" string?

My initial thoughts. What do you think?

cmarqu, 2016-02-01:
I think printing "Equality check failed!" only to negate that statement in surrounding text is confusing, or at least distracting when glancing over the log output.
That's why my user-specified check descriptions (i.e. the string that a testcase author supplies to all my check functions) are just stating *what* is being checked. The check function then adds text saying whether the check passed or failed.

Example output would be
```
Pass    check_equal(): Checking 'First error counter' actual=0 expected=0
ERROR   check_equal(): Checking 'Second error counter' actual=1 expected=0 failed!
```
or
```
Pass   cmpUnsigned() 'ADC conversion value' actual=32256 expected=32255 error=1 (0.002% FS) maxerror=2621
```

As you can see we have "Pass" as an extra "log level", yes. That makes it also easier to postprocess the logs in order to add ANSI color sequences.

LarsAsplund, 2016-02-03:
@cmarqu You're right, these messages are meant to be read so they should be as readable as possible. I see a number of different cases.

* For checks like `check(en = '1', "Checking that enable is active.")` the user is in full control of the message and making sure that it becomes readable in both the pass and fail cases.
* For some checks there are actually several checks being done internally.  For example, `check_stable` checks the stability of a `std_logic[_vector]` signal between a `std_logic` start event and a `std_logic` end event. In addition to checking the stability of the signal it will also check that the event signals doesn't have X values. If that is the case it will fail with something like `"Unknown start event"`. Such a message can be changed to become readable in the pass case as well. However, I'm thinking that these types of pass messages would make the log less clean for your use case, it would be better to just output _one_ pass message for the internal check that does the actual stability check. That check uses the user provided message and the user has to make sure it becomes readable.
* In case VUnit generates a predefined error message it also let you specify your own message to be appended. So `check_equal(en, '1', "Checking that enable is active.")` would today give

   ```
ERROR: Equality check failed! Got 0. Expected 1. Checking that enable is active.
```

   If the user is responsible for writing good messages according to the first bullet, and if some messages are excluded in the pass case according to the second bullet, we might consider to make this the general approach. No internal messages are provided in the pass case, only the one provided by the user. So a passing equality check would give

   `PASS: Checking that enable is active.`

   In this case the got and expected information is known so it doesn't provide any valuable information about the context. The drawback is that you lose some context when not comparing with a known constant. I'm not sure this is the perfect solution but I'm looking for something that allow us to do this without compromising too much what we have for the most common use case of not logging pass messages.

LarsAsplund, 2016-02-24:
@cmarqu I did some prototyping on the acknowledge-on-passing-check branch. The example code starts at line 194 in https://github.com/VUnit/vunit/blob/acknowledge-on-passing-check/examples/vhdl/check/check_example.vhd. It works for VHDL 2008 (at least with ModelSim) and gives the following output on stdout

```
# INFO: To support debugging you can get a positive acknowledge on a passing check.
# INFO: The message returned is the message provided to the check but the log level will be PASS.
# ERROR: Checking some condition.
# PASS: Checking some condition.
# INFO: Some checks like check_stable can fail with the given message ("Checking stability." in this case)
# ERROR: Checking stability.
# INFO: but also with messages not provided by the user. This is because there are several internal checks, not only one for checking the stability.
# ERROR: Unknown data in window.
# ERROR: Unknown end event.
# INFO: However, there is only one pass message for every top-level check called by the user. The provided message is used for that.
# PASS: Checking stability.
# INFO: There are also checks that return error messages which are a combination of the provided messages and internally generated messages.
# ERROR: Equality check failed! Got 0. Expected 1. Checking that condition is true.
# INFO: Still, when they pass only the provided message is given.
# PASS: Checking that condition is true.
# INFO: Typically you would only have this debug info on file. Acknowledge is off by default.
# ERROR: Checking some condition.
# INFO: The PASS message is not shown on stdout```

LarsAsplund, 2016-10-27:
@cmarqu Time to revisit this and see if it can be closed. I wasn't very pleased with the solutions presented so far so I did some additional thinking. @kraigher You've only expressed performance concerns and can save some time by jumping to the final paragraph.

I think the on- and off-line use cases/requirements/concerns I've heard can be summarized like this.

1. Use cases:
	* Pass messages can be used to see what requirements that are fulfilled by the DUT - I think the test cases are much better suited for this and we already have pass/fail status for those.
	* Pass messages can be used as a quality metric to show the amount of testing actually performed - Maybe a bit stronger use case than the first but we can already read the number of checks being performed and that would give a similar but much more dense metric.
	* Pass messages can be used as a simple way to provide extra debug information about the state of a DUT prior to an error. This is the best use case according to me and puts focus on presenting whatever DUT context that the check is aware of.
2. It should be possible to enable this for an existing testbench without having to rewrite the checks already written.
3. The messages should read nicely.
4. The performance impact for those not using this feature must be insignificant.

Today a failing `check(some_boolean_expression)` results in

`ERROR: Check failed.`

We actually have an exclamation mark today but I rather have a period. We have other means to put emphasis on errors so there's no need to shout.

Just adding a new `PASS` level would result in

`PASS: Check failed.`

but that would violate requirement 3. We've discussed using a more suitable default message that reads well in both situations, for example

`ERROR: Checking boolean expression.`
`PASS: Checking boolean expression.`

That works because of the `level` log format but it wouldn't read nicely with `raw`. Most people would use `level` but I still think this dependency has a bad smell. I would prefer if the default "message" is a special tag, currently `"+."` for reasons that I will explain later. It's then up to the check to decide what to do with that but if we do

`ERROR: Check failed.`
`PASS: Check passed.`

we stay readable for all formats and we also maintain the current look and feel for default error messages.

Specifying a custom message would replace the default so `check(parity_ok, "Checking parity.")` would result in

`ERROR: Checking parity.`
`PASS: Checking parity.`

The way I think about the `+` in the default message is that it tells the check to add a prefix to the rest of the message so `check(parity_ok, "+ for parity.")` would result in

`ERROR: Check failed for parity.`
`PASS: Check passed for parity.`

This means that you can also create a custom message that reads well with the `raw` format.

Today a `check_equal(1, 0, "Checking my data.")` results in

`ERROR: Equality check failed! Got 1. Expected 0. Checking my data.`

but that is inconsistent with what been described above. I would like to change this such that the first sentence is the default that is replaced if a message is given by the user. So

`check_equal(1, 0)` gives `ERROR: Equality check failed - Got 1. Expected 0.`
`check_equal(1, 0, "Checking my data")` gives `ERROR: Checking my data - Got 1. Expected 0.`
`check_equal(1, 0, "+ for my data")` gives `ERROR: Equality check failed for my data - Got 1. Expected 0.`

I use `-` to put some extra emphasis on the context which was the focus for the primary use case. In case of a passing check the context can be reduced

`PASS: Equality check passed for my data - Got 1.`

In general the check message format is

\<prefix\>\<message\> - \<context\>

where each field may or may not exist depending on the situation.

I've made the changes described above for `check_equal` and performed some performance testing with our supported simulators. My test is based on checking the output of a 32-bit counter. That is a design with very much checking in relation to design size so the relative performance hit is not likely to be as high in reality. Three of the tested simulators had a penalty between -1.3% (performance gain) to 1.5% so it seems the penalty is insignificant to the variation in simulation times. The fourth simulator had a penalty of 3.5% corresponding to 2 extra seconds added to a 54 second run with 100 million checks. Where I work 100 million checks corresponds to ~100 frames of video processing which would take days or weeks to simulate. I don't think this is a problem.

What do you think? I this a reasonable approach?

LarsAsplund, 2016-10-28:
For comparison: One million pass messages saved to file takes an extra 2.5 minutes on my laptop. If such a  run corresponds to an hour of simulation which would be typical in my current project then the penalty is justified if you think the extra debug information can save you one simulation in every 25 or so. That seems to validate the use case.

cmarqu, 2016-10-28:
Yes, that all sounds very reasonable to me, supporting all use cases we determined.

LarsAsplund, 2016-10-28:
@cmarqu @joshrsmith Thanks for the review. I'll complete the remaining checks and then push the branch.

kraigher, 2016-10-28:
So if this is not enabled the cost will be just an if-statement where the branch is not taken?

LarsAsplund, 2016-10-28:
@kraigher There are actually two branches not taken. One if statement prevents the pass message from being generated if not enabled and the other prevents a log for the message to be issued. The only action on a passing check if pass messages are disabled is that the internal statistics are updated, +1 for the number of passing checks. This is the same behavior we have today.

kraigher, 2016-10-28:
ok sounds reasonable.

LarsAsplund, 2017-01-06:
I've push this update now. The only real difference from what being discussed before is that I've hidden the `+` sign. Actually it's not even a `+` anymore. Now you have a `result` function instead.

``` vhdl
check_equal(1, 0, result("for my data"))
``` 
gives

``` console
ERROR: Equality check failed for my data - Got 1. Expected 0.
```
or

``` console
PASS: Equality check passed for my data - Got 1.
```

closed 2017-01-06 by LarsAsplund