VUnit/vunit #55: Generate check_equal and to_string for records within package.

kraigher, 2015-07-01:
We should do this similarly to the code generation for com. I think this case is the most common and useful one for code generation.

In run.py it could look like:
```
pkg.generate_checks(file_name=...)
```

I think the user should optionally be able to decide were the file is placed.

LarsAsplund, 2015-07-01:
Yes, that seems like a reasonable enhancement now that we have the com code to build upon. I guess that we should expose the "=" function as well if we expose to_string. We should also look at the possibility of adding other relations as well. Equal will be the most commonly used relation but I guess others like not_equal would come almost for free.

kraigher, 2015-07-01:
"=" is already automatically defined for new types in vhdl.

kraigher, 2015-07-01:
Actually I think the run.py code should look like this:
```
from vunit.codegen import generate_checks
generate_checks(pkg, file_name="...")
```

I think the code generation for com should be converted to follow a similar style. That keeps them separate from the main API and prevents the main API from growing to many methods.