VUnit/vunit #56: user_guide example on linux/modelsim

rndhro, 2015-07-06:
Hi,

there's probably something wrong with my environment but I can't spot it - can you give advice?

```
$ python run.py lib.tb_example*
Starting lib.tb_example
# Command 'vsim  -modelsimini /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/modelsim/modelsim.ini -wlf /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/tests/lib.tb_example/msim/vsim.wlf -quiet -t ps -grunner_cfg="enabled_test_cases : None,output path : /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/tests/lib.tb_example/,active python runner : true" lib.tb_example(tb) ' failed 
# Bad flag from vsim_extra_args? 
fail (P=0 S=0 F=1 T=3) lib.tb_example (0.5 seconds)

Starting lib.tb_example_many.test_pass
# Command 'vsim  -modelsimini /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/modelsim/modelsim.ini -wlf /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/tests/lib.tb_example_many.test_pass/msim/vsim.wlf -quiet -t ps -grunner_cfg="enabled_test_cases : test_pass,output path : /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/tests/lib.tb_example_many.test_pass/,active python runner : true" lib.tb_example_many(tb) ' failed 
# Bad flag from vsim_extra_args? 
fail (P=0 S=0 F=2 T=3) lib.tb_example_many.test_pass (0.0 seconds)

Starting lib.tb_example_many.test_fail
# Command 'vsim  -modelsimini /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/modelsim/modelsim.ini -wlf /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/tests/lib.tb_example_many.test_fail/msim/vsim.wlf -quiet -t ps -grunner_cfg="enabled_test_cases : test_fail,output path : /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/tests/lib.tb_example_many.test_fail/,active python runner : true" lib.tb_example_many(tb) ' failed 
# Bad flag from vsim_extra_args? 
fail (P=0 S=0 F=3 T=3) lib.tb_example_many.test_fail (0.0 seconds)

==== Summary =========================================
fail lib.tb_example                (0.5 seconds)
fail lib.tb_example_many.test_pass (0.0 seconds)
fail lib.tb_example_many.test_fail (0.0 seconds)
======================================================
pass 0 of 3
fail 3 of 3
======================================================
Total time was 0.5 seconds
Elapsed time was 1.8 seconds
======================================================
Some failed!
```

```
$ env | grep MODEL
MODEL_VERSION=10.1c_1
MODEL_TECH=/mnt/data/tools/Modelsim/10.1c_1/modeltech/linux_x86_64
MODEL_SIM_ROOT=/mnt/data/tools/Modelsim/10.1c_1/modeltech
MODEL_SIM_PATH=/mnt/data/tools/Modelsim/10.1c_1/modeltech/linux_x86_64
```

```
$ vsim -version
Model Technology ModelSim SE-64 vsim 10.1c Simulator 2012.07 Jul 27 2012
```

kraigher, 2015-07-06:
Could you try to open a modelsim gui and run one of the vsim commands above to see what it says?

kraigher, 2015-07-06:
More specifically could you open a modelsim gui and issue the following commands in the same working directory as you ran run.py.

```tcl
set vsim_extra_args ""
vsim {*}${vsim_extra_args} -modelsimini vunit_out/modelsim/modelsim.ini -wlf -quiet -t ps  -grunner_cfg="" lib.tb_example(tb) 
```

kraigher, 2015-07-06:
You could also do `run.py --gui=load` which will launch a modelsim gui and see if there is more info about the error in red text in the transcript window..

rndhro, 2015-07-07:
run.py --gui=load: vsim opens up, also prints:
```
# do /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/tests/lib.tb_example/msim/gui_load.do 
# Command 'vsim  -modelsimini /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/modelsim/modelsim.ini -wlf /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/tests/lib.tb_example/msim/vsim.wlf -quiet -t ps -grunner_cfg="enabled_test_cases : None,output path : /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/tests/lib.tb_example/,active python runner : true" lib.tb_example(tb) ' failed 
# Bad flag from vsim_extra_args? 
```
(unrelated: after closing vsim, each test is shown as 'passed')

from vsim gui:
```
set vsim_extra_args ""
vsim ${vsim_extra_args} -modelsimini vunit_out/modelsim/modelsim.ini -wlf -quiet -t ps  -grunner_cfg="" lib.tb_example(tb)
# vsim -modelsimini vunit_out/modelsim/modelsim.ini -t ps -wlf -quiet -grunner_cfg=\"\" lib.tb_example(tb) 
# ** Note: (vsim-3812) Design is being optimized...
# Loading std.standard
# Loading std.textio(body)
# Loading vunit_lib.lang(body)
# Loading vunit_lib.textio(body)
# Loading ieee.std_logic_1164(body)
# Loading ieee.numeric_std(body)
# Loading vunit_lib.string_ops(body)
# Loading vunit_lib.log_types_pkg(body)
# Loading vunit_lib.log_formatting_pkg(body)
# Loading vunit_lib.log_special_types_pkg(body)
# Loading vunit_lib.log_base_pkg(body)
# Loading vunit_lib.log_pkg(body)
# Loading vunit_lib.check_types_pkg(body)
# Loading vunit_lib.check_special_types_pkg(body)
# Loading vunit_lib.check_base_pkg(body)
# Loading vunit_lib.check_pkg(body)
# Loading vunit_lib.dictionary(body)
# Loading vunit_lib.path(body)
# Loading vunit_lib.run_types_pkg(body)
# Loading vunit_lib.run_special_types_pkg(body)
# Loading vunit_lib.run_base_pkg(body)
# Loading std.env(body)
# Loading vunit_lib.vunit_stop_pkg(body)
# Loading vunit_lib.run_pkg(body)
# Loading work.tb_example(tb)#1
```
so this looks good... note that I had to remove the "{*}" to make it work.

this time with the vsim command from the error string:
```
vsim  -modelsimini /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/modelsim/modelsim.ini -wlf /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/tests/lib.tb_example/msim/vsim.wlf -quiet -t ps -grunner_cfg="enabled_test_cases : None,output path : /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/tests/lib.tb_example/,active python runner : true" lib.tb_example(tb)
# vsim -modelsimini /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/modelsim/modelsim.ini -quiet -t ps -wlf /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/tests/lib.tb_example/msim/vsim.wlf {-grunner_cfg=enabled_test_cases : None,output path : /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/tests/lib.tb_example/,active python runner : true} lib.tb_example(tb) 
# ** Warning: (vsim-3351) Invalid value "enabled_test_cases : None,output path : /mnt/data/projects/vunit/examples/vhdl/user_guide/vunit_out/tests/lib.tb_example/,active python runner : true" for generic "runner_cfg".  Not using this value.
#    Time: 0 ps  Iteration: 0  Instance: /tb_example File: /mnt/data/projects/vunit/examples/vhdl/user_guide/tb_example.vhd
# ** Fatal: (vsim-3350) Generic "runner_cfg" has not been given a value.
#    Time: 0 ps  Iteration: 0  Instance: /tb_example File: /mnt/data/projects/vunit/examples/vhdl/user_guide/tb_example.vhd Line: 11
# FATAL ERROR while loading design
# Error loading design
```

kraigher, 2015-07-07:
The {\*} TCL operator was something that we recently introduced. Without it the `vsim_extra_args` parameter to `vunit_load` cannot contain multiple separate flags. The operator splits a variable into multiple arguments before calling the parent command. I was worried that it might cause problems since it is a newer TCL feature. I removed the {\*} operator in the `issue56` branch using `eval` instead. Could you try that?

Also the reason it says passed when you close the GUI is that the exit code from you closing the GUI manually is 0. Had you written `quit -f -code 1` the test would have failed. The GUI mode is only intended to be a quick way to open **a** test case in the GUI with everthing pre-loaded for some manual debugging so the test status and continued `run.py` execution afterwards is not seen as interesting. Maybe it could be improved if it confuses someone. A failure such as a failing assert can be clearly seen in the GUI printout when running.

rndhro, 2015-07-07:
branch issue56 works!
Modelsim 10.1 comes with tcl 8.4 (if this is of any help...)

On vsim gui closing: indeed, I closed it with quit -f so return code is 0 - thanks for the clarification.

kraigher, 2015-07-07:
I have merged this to master and made a new release. Thank you for reporting.

closed 2015-07-07 by kraigher

PingNie, 2017-12-19:
@kraigher I am trying to use vunit. I tried to run user_guide/run.py and faced a Error: (volt - 2253) No module name found in command-line "true, enabled_test_cases".   #Error loading design. when I tried python run.py --gui. It showed the same errors in the modelsim's vsim.   My modelsim version is 10.1a on windows 7. thank you~

kraigher, 2017-12-19:
@PingNie 10.1a is probably to old and not compatible with VUnit. The oldest version known to work is 10.1b I believe. Of course it would be theoretically possible for us to support 10.1a but I do not have access to it. You have to upgrade your simulator. There is the free Intel Modelsim starter edition (former Altera) or GHDL. The other alternative is that you try to make a fix for 10.1a yourself, in that case have a look in vunit/modelsim_interface.py and vunit/vsim_simulator_mixin.py