VUnit/vunit #57: Support for Modelsim vlog include dirs

rndhro, 2015-07-08:
Verilog include directories have to be specified with ```+incdir+[dirname]``` in the ```vlog``` command - are there plans to support this?
For the moment I modified the call to ```vlog``` in modelsim_interface.py for my needs, but this is obviously not portable. However it seems a clean implementation requires some work...
Maybe providing the ```config``` object already to ```compile_project()``` and having something like ```vu.set_sim_option("vlog_extra_args.gui","+incdir+dirname")```?

kraigher, 2015-07-08:
We are working on adding full support for SystemVerilog in the systemverilog branch, issue #49. The goal is to make VUnit a testing tool also for SystemVerilog projects. In this branch we have already added support for setting include directories. It is still a work in progress though.

kraigher, 2015-07-08:
In the systemverilog branch the add_source_files method has an include_dirs argument. Remember that this is not official and still a work in progress.

rndhro, 2015-07-08:
Thanks for the pointer, I'll give it a try. Having include_dirs on a per-file basis is indeed better than on project level!

closed 2015-07-08 by rndhro