VUnit/vunit #58: vhdl_parser depends on coding style to find ports/generics

rndhro, 2015-07-08:
works:
```vhdl
entity tb_example_many is
  generic (runner_cfg : runner_cfg_t);
end entity;
```
works:
```vhdl
entity tb_example_many is
  generic 
  (
    runner_cfg : runner_cfg_t
  );
end entity;
```

**does not work**:
```vhdl
entity tb_example_many is generic
  (
    runner_cfg : runner_cfg_t
  );
end entity;
```

VHDL does not require a newline before ```generic``` or ```port``` but vhdl_parser.py:387 /:413 currently implement it this way.

kraigher, 2015-07-08:
I have fixed the problem and added some other corner case tests.

kraigher, 2015-07-08:
Some time in the future I plan to rewrite the VHDL parser to be based on tokens by first tokenizing the input instead of the current raw regex approach. This will scale better and be more maintainable. The Verilog parser that I started working on in the systemverilog branch is already token based.

closed 2015-07-08 by kraigher

rndhro, 2015-07-08:
thanks for the fix - that was fast...again!
Parsing VHDL can be really tricky...I think there are a few more possible pitfalls in the current implementation, but in 99% of the cases probably not worth the effort...

kraigher, 2015-07-08:
Yes the current raw regexp implementation can be tricked by cleverly crafted input as it is a weak method of parsing. That is why I want to rewrite it to use proper tokenization. It has worked well in practice for the parsing needs of VUnit though.