VUnit/vunit #59: Parsing Error

amartinray, 2015-07-15:
When I try to compile my library, I get this error:

Traceback (most recent call last):
  File "/home/user/path/scripts/vunit-master/vunit/project.py", line 379, in __init__
    design_file = vhdl_parser.parse(code, name)
  File "/home/user/path/scripts/vunit-master/vunit/vhdl_parser.py", line 60, in parse
    design_file = VHDLDesignFile.parse(code)
  File "/home/user/path/scripts/vunit-master/vunit/vhdl_parser.py", line 95, in parse
    packages=list(VHDLPackage.find(code)),
  File "/home/user/path/scripts/vunit-master/vunit/vhdl_parser.py", line 245, in find
    yield cls.parse(sub_code[:match.end()])
  File "/home/user/path/scripts/vunit-master/vunit/vhdl_parser.py", line 254, in parse
    constant_declarations = cls._find_constant_declarations(code)
  File "/home/user/path/scripts/vunit-master/vunit/vhdl_parser.py", line 280, in _find_constant_declarations
    constant_declarations.append(VHDLConstantDeclaration.parse(sub_code))
  File "/home/user/path/scripts/vunit-master/vunit/vhdl_parser.py", line 555, in parse
    subtype_indication = VHDLSubtypeIndication.parse(sub_code[:expression_start].strip())
  File "/home/user/path/scripts/vunit-master/vunit/vhdl_parser.py", line 511, in parse
    type_mark = subtype_indication_declaration.group('type_mark')
AttributeError: 'NoneType' object has no attribute 'group'    
ERROR - Failed to parse /path/to/package_file.vhd                                  

I'm using the latest master build.  I was able to isolate it to these lines of my vhdl package (names changed to protect the innocent):

  type  MyType      is (MY_ONE,  MY_TWO, MY_THREE, MY_FOUR);
  type  MyArrayType is array (MyType'left to
                                     MyType'right) of natural;
  constant My_STUFF : ---------------------------------------------------
    MyArrayType := (MY_ONE      =>  1, 
                           MY_TWO => 2,
                           MY_THREE      => 3,
                           MY_FOUR  => 4);

If those lines are included in my package, I get the error. If I comment them out, it is able to continue to the compile.

Any idea on what the problem is?  

Thanks.

kraigher, 2015-07-16:
Should be fixed in release v0.26.0

kraigher, 2015-07-16:
Thank you for reporting.

closed 2015-07-16 by kraigher