VUnit/vunit #60: Change VHDL Standard Per Library

amartinray, 2015-07-16:
Loving this tool so far.  I have a question or request if it doesn't exist.

My design has multiple libraries.  Most require VHDL-2008 but one requires VHDL-1993.  Is there a way to specify which to use on a per library basis?  

Thanks.

kraigher, 2015-07-16:
It is not possible to individually set the VHDL standard. It is globally set. We have in some early non-public version allowed users to set it per file but we later simplified it. While it would be technically possible for us to allow individually setting the VHDL standard without much work I would first like to understand if it is necessary. My belief is that there is no good reason to not compile everything with VHDL 2008 for simulation. The only non-backwards compatible issue with 2008 is if an identifier collides with one of the new keywords. Also simulators such as GHDL either do not allow mixing the standard. ModelSim does not complain but can crash due to inconsistency in my personal experience. I am thus curious to know the reason why you cannot compile this library with 2008? If it is truly blocking you I could try to prioritise supporting individually setting the VHDL standard.

Also if you like the tool you can boost its reputation by giving it a GitHub star in the upper right corner of the website.

amartinray, 2015-07-16:
Thanks for the reply.  I have an IP library that I don't have control over that uses the "default" reserved word in one of it's files.  They have updated their library and will be getting me a fix soon but I don't have it yet.  If this request is a hassle, then spend your time in other areas.  I'll get the fix from the IP creator.  It is not worth changing your priorities.  

I've given a star.

kraigher, 2015-07-16:
Ok based on this information I will put this on the waitlist for now. Thank you for your feedback.

closed 2015-11-28 by kraigher