VUnit/vunit #62: simulations with SDF

vermaete, 2015-07-27:
For doing postlayout simulations with Modelsim I'm using the 'set_sim_option' to pass the sdf file.

E.g.:
vu.set_sim_option("vsim_extra_args", "-sdfmax /dut=../implementation/libero/designer/{0}/{0}_ba.sdf".format(COMPONENT))

It's working fine.  But should there not be a solution that is simulator independent?

kraigher, 2015-07-27:
The vsim_extra_args options exists so that we do not have to create a simulator independent interface for every conceivable type of option that is similar among simulators. Just doing `vsim -h | grep sdf` shows many different SDF related options:
```console
vsim -h | grep sdf
   -sdfmax[@<delayScale>] [<instance>=]<sdffile>
                           timing from <sdffile>, scaled by <delayScale>
   -sdfmaxerrors <n>       Max number of missing instances reported (default is 5)
   -sdfmin[@<delayScale>] [<instance>=]<sdffile>
                           timing from <sdffile>, scaled by <delayScale>
   -sdfminr[@<delayScale>] [<instance>=]<sdffile>
   -sdfnoerror             Treat SDF errors as warnings
   -sdfnowarn              Disable warnings from SDF annotator
   -sdfreport=<fileName>   Report unannotated/partially-annotated specify objects into <fileName>
   +sdf_report_unannotated_insts Enable error messages for any un-annotated Verilog instances with
   -sdftyp[@<delayScale>] [<instance>=]<sdffile>
                           timing from <sdffile>, scaled by <delayScale>
   +sdf_verbose            Display SDF annotator status messages
   +nosdferror             Treat SDF errors as warnings
   +nosdfwarn              Disable warnings from SDF annotator
   +sdf_iopath_to_prim_ok  Prevent vsim from issuing an error when it cannot locate specify path 
   +sdf_nocheck_celltype   Disable check between SDF celltype name and
```
There is a balance between if something should have its own simulator independent interface or if the user has to provide the simulator specific flags themselves. In my opinion the extra weight of making a Python SDF configuration interface part of vunit is not worth is since the vsim_extra_args methods already works. vsim_extra_args was created for such specific configuration.

vermaete, 2015-07-28:
Thanks for the answer.  My goal was to spare our limited number of Modelsim licenses and keep them for development and debugging only.  And use the reborn ghdl for massive regression runs and CI. Not having any compiler/simulator specific code in the 'run.py' makes this a bit easier for the user.  But I do understand that having everything generic is not possible.

kraigher, 2015-07-28:
Sounds like a good approach, GHDL has become very good lately. Do you also want to use SDF when simulating with GHDL? We have no similar hook to vsim_extra_args for GHDL currently.
You can set the vsim_extra_args hook regardless of what simulator you are using, when using GHDL it will be ignored.

vermaete, 2015-07-28:
ghdl is a plan, but not yet planned :-)  So, at the moment we don't use it
yet.

kraigher, 2015-07-28:
I would recommend the LLVM backend of GHDL. It is easy to build from source and runs fast. The GHDL folks are planning on making a binary release soon. The current binary release is quite old and lacks many improvements and bug fixes required to work with VUnit. That is why you have to build from source at the moment.

closed 2015-07-28 by vermaete