VUnit/vunit #63: set the output_path

vermaete, 2015-07-29:
When setting the output_path like this:
vu = VUnit.from_argv(argv=["--output-path=vunit_out_postlayout"])

It's working but I'm not longer able to do things like:
python run.py --gui load

So, I specify the output_path now:
vu._output_path="vunit_out_postsynth"

But that's not the most beautiful thing to do (a protected member in Python).

Is there a 'setter' for the output_path member?

kraigher, 2015-07-29:
The `argv` argument of the `from_argv` method overrides the `argv` provided from the command line and was added to simplify testing of VUnit itself. It will not be possible to set the `output_path` on an already initialized instance of VUnit since it performs operations on the `output_path` at initialization. 

What you can do is use the `from_args` method together with the `VUnitCLI` class to change the default value of the `output_path` flag:
```python
from vunit import VUnitCLI, VUnit
cli = VUnitCLI()
# Change the default value of 'output_path'
cli.parser.set_defaults(output_path=join(dirname(__file__), "vunit_out_postlayout"))
args = cli.parse_args()
vu = VUnit.from_args(args=args)
``` 

**NOTE:** The `join(dirname(__file__), ...)` is to ensure the path is always fixed relative to the location of `run.py` rather than depend on where it was executed.

vermaete, 2015-07-29:
works perfect, thanks

closed 2015-07-29 by vermaete