VUnit/vunit #73: Unhandled exception, KeyError, when referencing non-existant architecture in a configuration

joshrsmith, 2015-09-15:
Not sure what the intended functionality is for Vunit, but consider the following case.

```
entity foo is
end entity foo;

architecture foo_arch of foo is
begin

end architecture foo_arch;
```

```
entity foo_tb is
end entity foo_tb;

architecture foo_tb_arch of foo_tb is
   component foo is
   end component foo;
begin
   uut : foo;
end architecture foo_tb_arch;

configuration foo_tb_cfg of foo_tb is
   for foo_tb_arch
      for uut : foo
         use entity work.foo(foo2_arch);  -- NOTE this is foo2_arch and not foo_arch
      end for;
   end for;
end configuration foo_tb_cfg;
```

In the foo_tb.vhd file, foo(foo2_arch) is called out instead of foo_arch. But foo(foo2_arch) has not been analyzed anywhere. The result is a dictionary KeyError.

The stack trace is similar to follows:
```
  File "e:\vunit\vunit\ui.py", line 506, in get_project_compile_order
    return self._project.get_dependencies_in_compile_order(target=target)
  File "e:\vunit\vunit\project.py", line 284, in get_dependencies_in_compile_order
    return self.get_files_in_compile_order(incremental=False)
  File "e:\vunit\vunit\project.py", line 260, in get_files_in_compile_order
    dependency_graph = self._create_dependency_graph()
  File "e:\vunit\vunit\project.py", line 240, in _create_dependency_graph
    add_dependencies(self._find_other_design_unit_dependencies, vhdl_files)
  File "e:\vunit\vunit\project.py", line 230, in add_dependencies
    for dependency in dependency_function(source_file):
  File "e:\vunit\vunit\project.py", line 154, in _find_other_design_unit_dependencies
    file_name = architectures[name]
KeyError: u'foo2_arch'
```

The work-around is to not attempt to reference/use an architecture that has not been analyzed.
My question is -- should this be logged as an error/warning, and continue execution, instead of an unhandled exception which terminates execution/analysis? There seems to be other cases (such as a library that cannot be found) which generates warnings instead of an unhandled exception -- why not this?

kraigher, 2015-09-15:
Yes it would be better if it was a warning and not an unhandled exception. I will fix it.

kraigher, 2015-09-15:
Fixed and released in v0.32.0.

closed 2015-09-15 by kraigher