VUnit/vunit #74: False warning generated for use clause: use libraryname.all

joshrsmith, 2015-09-15:
Consider the following file, compiled into the testcase library. Assume that `foo` has also been compiled into the testcase library.

```
library testcase;
use testcase.all;

entity foo_tb is
end entity foo_tb;

architecture foo_tb_arch of foo_tb is
begin
   uut : entity foo;
end architecture foo_tb_arch;
```

The following warning is falsely issued:
```
WARNING - (filename): failed to find a primary design unit 'all' in library 'testcase'.
```

`all` is not a design unit, but rather it is bringing all design units in testcase library into scope. In this case you can directly reference `entity foo` instead of `entity testcase.foo` or `entity work.foo`

kraigher, 2015-09-16:
Yes this case is missing from our dependency scanner. The case is a bit trickier since it requires deeper analysis to figure out the implicit dependencies that `foo` was an entity in the `testcase` library. I also suspect there are more complex cases when packages are used implying we need to parse every identifier in the file to see if it was a `packagename.name` where `packagename` was exposed by `use libraryname.all`. But if we ignore the package case the entity case is achievable without to much work.

kraigher, 2015-09-28:
Supporting this in a non ugly way would require some refactoring to our current parser. It has some limitations which we run into for this case. I would like to support this eventually but I estimate it will take some time. I would therefore recommend not using `use libraryname.all` for now. I hope it is not that big of a deal since it is trivial to re-write the VHDL code to not run into this limitation.

joshrsmith, 2015-09-28:
Understood -- I typically don't use it, and the work-around is
straightforward.
There are a few spots in the code base that I work on that do it, which is
why I discovered this bug/limitation in the first place.

LarsAsplund, 2016-06-10:
All issues that can be solved with a more advanced VHDL parser will be tracked in #174 .

closed 2016-06-10 by LarsAsplund