VUnit/vunit #76: ImportError: No module named 'vunit.parsing'

vermaete, 2015-09-22:
after a 'git pull' (commit c7bcae20df4b88bce9bb722e722a1baebfa13c03) I got the following error when running the 'run.py' script.

    python run.py --compile
    Traceback (most recent call last):
      File "run.py", line 3, in <module>
        from vunit import VUnit
      File "C:\Python34\lib\site-packages\vunit\__init__.py", line 11, in <module>
        from vunit.ui import VUnit
      File "C:\Python34\lib\site-packages\vunit\ui.py", line 28, in <module>
        from vunit.project import Project, file_type_of
      File "C:\Python34\lib\site-packages\vunit\project.py", line 22, in <module>
        from vunit.parsing.verilog.parser import VerilogParser
    ImportError: No module named 'vunit.parsing'

It did not update for a few days/weeks.  So I don't know when this happened.

Grepping in the Python installation path for VerilogParser is giving me:

    Binary file __pycache__/project.cpython-34.pyc matches
    project.py:from vunit.parsing.verilog.parser import VerilogParser
    project.py:                 verilog_parser=VerilogParser()):
    test/unit/test_verilog_parser.py:class TestVerilogParser(TestCase):  # pylint: disable=too-many-public-methods
    Binary file test/unit/__pycache__/test_verilog_parser.cpython-34.pyc matches

My guess is something is missing the the config.py script.  Or I did something stupid....

kraigher, 2015-09-22:
Very likely just an omission in setup.py.

vermaete, 2015-09-22:
directory 'vunit/parser' is missing in setup.py ?

vermaete, 2015-09-22:
Removing the vunit from the Python installation path (c:\Python34\Lib\site-packages) and setting the PYTHONPATH to the git clone directory did the trick.  So: yes, it is the setup.py script.

kraigher, 2015-09-22:
Yes I was immediately convinced it was the setup.py file. Neither me nor Lars really use the setup.py file and we perform quite exhaustive testing so such an error could not come from anywhere else. We should do some testing also with the setup.py file when adding new modules in the future.

commit dab64f4c (2015-09-22): Fixed incomplete setup.py. Closes #76.
closed 2015-09-22 by kraigher