VUnit/vunit #77: Generics capitalization issue

joshrsmith, 2015-10-02:
I am using Riviera/Active HDL. Not sure if this affects other simulators.

When generating testcases for a testbench by assigning generic values, if the generic has capitals in it, the generic will not be assigned to the simulation successfully.

For example, if the "generate_tests" example is modified such that:
```
entity tb_generated is
  generic (
    runner_cfg : runner_cfg_t := runner_cfg_default;
    output_path : string;
    Data_width : natural;        -- !! NOTE capital D
    sign : boolean;
    message : string);
end entity;

```
and the run.py is updated accordingly, the simulator will complain that "No value assigned to generic/parameter "Data_width""

When looking at the batch files generated for running the simulation, it appears that the generics are not present in there, either:
```
proc vunit_load {} {
    set vsim_failed [catch {
        vsim  -g/tb_generated/message=set-for-entity -g/tb_generated/runner_cfg="enabled_test_cases : Test 1,output path : E::/vunit/vunit_out/tests/lib.tb_generated.data_width=1,,sign=False.Test 1/,active python runner : true" -g/tb_generated/output_path="E:/vunit/vunit_out/tests/lib.tb_generated.data_width=1,sign=False.Test 1/" -g/tb_generated/sign=False -lib lib tb_generated a
    }]
    if {${vsim_failed}} {
        return 1
    }

    set no_vhdl_test_runner_exit [catch {examine /vunit_lib.run_base_pkg/runner.exit_simulation}]
    set no_verilog_test_runner_exit [catch {examine /\\package vunit_lib.vunit_pkg\\/__runner__}]
    if {${no_vhdl_test_runner_exit} && ${no_verilog_test_runner_exit}}  {
        echo {Error: No vunit test runner package used}
        return 1
    }
    return 0
}
```

I am guessing this is some edge case related to case-sensitivity in VHDL.

kraigher, 2015-10-02:
@joshrsmith I can reproduce the problem and it is not simulator specific. 

The VUnit parser converts everything to lower case since VHDL is not case sensitive. Thus a Data_width generic is called data_width by VUnit internally. The add_config and set_generic methods of the VUnit Python class however does not transform the generic name provided by the user into lower case when checking against the internal representation causing the problem. If you would reference the generic as entirely lower case in your `run.py` file you could work around the problem.

I will fix this right away and make a new release.

joshrsmith, 2015-10-02:
The work-around you suggested worked for me.

A fix would be appreciated in order to avoid this issue in the future--I thought I was losing my mind with an example nearly identical to the one provided with vunit.

kraigher, 2015-10-02:
I will push a fix just about any minute now. I also added a warning message when the user specifies a generic that is not present in the entity so that would have triggered in this case and made the VUnit bug easier to locate.

kraigher, 2015-10-02:
Fixed and released in v0.38.0. Thank you for reporting it.

closed 2015-10-02 by kraigher

joshrsmith, 2015-10-02:
Confirm as fixed.

commit 8360eaab (2015-10-02): Lower case all generics from the user to match internal representation. Fixes #77