VUnit/vunit #78: Compiling source files from a single directory to two separate libraries

danielblomkvistbitvis, 2015-10-15:
Hi,

I'm attempting to compile source files from a single directory into two separate libraries. I've added the source files in my run.py, but during compilation only one of the two libraries are successfully compiled.

The directory structure is as follows:

ModuleA:
- ModuleAFile1.vhd
- ModuleAFile2.vhd
- CommonFile1.vhd
- CommonFile2.vhd

ModuleB:
- ModuleBFile1.vhd

ModuleC:
- ModuleCFile1.vhd

An this is the library structure I want:

LibraryA:
- ModuleAFile1.vhd
- ModuleAFile2.vhd

LibraryB:
- ModuleBFile1.vhd
- CommonFile1.vhd
- CommonFile2.vhd

LibraryC:
- ModuleCFile1.vhd
- CommonFile1.vhd
- CommonFile2.vhd

This fails as LibraryB can't find the packages that reside in CommonFile1.vhd and CommonFile2.vhd in its work library during compilation.

I have discovered that if  the ordering of my calls to add_source_files() in the run.py script has an effect on which of LibraryB and LibraryC that won't compile. 

In my example here LibraryC will compile, while LibraryB will fail. Now if I switch the ordering so that LibraryB has its calls to add_source_files() after LibraryC, LibraryC will be the one that fails.

From running with --log-level info I can see that it seems like the source files CommonFile1.vhd and CommonFile2.vhd are added to both libraries, but maybe the second one overwrites the first?

It seems to me like dependencies are set up correctly w.r.t. file location.

kraigher, 2015-10-15:
Yes this is a bug. I will hopefully fix it this evening. Thank you for reporting.

kraigher, 2015-10-15:
Fixed and released in v0.39.0.

commit 7d2540e9 (2015-10-15): Fixes #78. Compiling the same file into multiple libraries did not work.
closed 2015-10-15 by kraigher