VUnit/vunit #79: VUnit support for code coverage

jonasalowersson, 2015-10-19:
A few additions are needed in order to make it possible to collect code coverage in Modelsim.
- Modelsim specific flag --coverage to enable the features below
- coverage arguments to vcom and vsim
- "coverage save" command in the scripts that are run in Modelsim
- merge of coverage databases from all test cases after finished simulation

vermaete, 2015-10-19:
When using a Modelsim that does not support coverage (e.g. OEM versions) should not set those coverage options.  They would fail the simulations.  

Although it would be good to be able to run the simulation with the same python scripts (run.py) with two kind of Modelsims (a version with coverage and a version without coverage).  E.g. When running at the CI build server, the full blown Modelsim is selected.  When running at a desktop PC to debug a testcase, the 'light' version can be used.  But both cases are starting with the same repository from VC (e.g. Git) without modifications (e.g. related to coverage or not).

LarsAsplund, 2015-10-19:
From time to time we get requests for features not supported by the free licenses we use to do the ModelSim integration. We recently got a partner type of license from Aldec and I think it's time to see if we can have something similar from Mentor Graphics. I found something called the OpenDoor program but I'm not sure if that is what we're looking for. Anyway, that program requires that we have customer references so I might get back to you on that.

kraigher, 2015-10-19:
@jonasalowersson I know you have already done some work on this. Could you publish a Fork with those initial changes so that I can review them and figure out the best way to integrate them?

@vermaete Yes the coverage specific flags/tcl should only be present when the `--coverage` flag is set. We should not assume all users have coverage licenses. It is possible to run the same run.py file with two different kinds of Modelsim. Just set the PATH environment variable. The PATH environment variable can be set from Python.

kraigher, 2015-10-21:
@jonasalowersson I have made some changes to your commit that I thought was necessary. I pushed the changes to the [codecov](https://github.com/LarsAsplund/vunit/tree/codecov) branch in the main VUnit repostory. Could you check it out and see that it still works?

jonasalowersson, 2015-10-22:
It works fine. 
However, I have noticed that Modelsim seems to collect all coverage types ('bcestf') that was specified during compile even if only a subset of them are specified for simulation. Therefore, I have added a few lines to include the coverage types also in the compile functions, and committed this to my fork after updating it to reflect your changes.
Please also note that I have not been able to verify the functionality for verilog files, since I don't have access to a Modelsim version with support for mixed language.

kraigher, 2015-10-22:
@jonasalowersson I have had a look at your changes and added some myself in the codecov branch of the main VUnit repository. If you are happy with this and it works I am ready to integrate it into the VUnit master branch.

jonasalowersson, 2015-10-23:
I have made some further tests, and found a problem, which was there already in my first commit and has not gone away. The coverage.ucdb file from the last simulation has not yet been written when the merge operation begins. It looks like if _run_test(...) at line 358 in ui.py returns as soon as the last vsim process is started, not when it is finished. Somehow the start of post_process() must be delayed until that last vsim process is finished also.

kraigher, 2015-10-23:
I can have a look at this. You should be able to use --new-vsim flag to run a new vsim process for each test as a work arround.

jonasalowersson, 2015-10-23:
I tried the workaround, and it solves the problem, but of course it would be better if you could find a way to make it work without the workaround.

vermaete, 2015-10-23:
Having default options to the coverage argument?
--coverage [options]

kraigher, 2015-10-25:
@vermaete I have fixed it such that everything is enabled when using the flag without any argument.

kraigher, 2015-10-25:
@jonasalowersson I have pushed a fix that will probably solve your problem of missing ucdb-files without running with `--new-vsim`. I have also added a warning when an ucdb file is missing. Could you try it?

jonasalowersson, 2015-10-26:
I have tried your latest commit, and it works fine, thanks! As far as I'm concerned, this functionality is sufficient, and I would be happy if you could integrate it in the VUnit master branch.

kraigher, 2015-10-26:
I have integrated it into master and released 0.42.0

closed 2015-10-26 by kraigher