VUnit/vunit #80: A workaround for a bug in the Modelsim vmap command on some Linux hosts

danielblomkvistbitvis, 2015-10-19:
Hi,

There is a bug in the Modelsim 'vmap' command that results in Modelsim incorrectly reporting that the 'modelsim.ini' file has been successfully modified, when in fact 'modelsim.ini' has not been modified. The result is compilation failure. 

This bug only occurs on some Linux hosts. Specifically, I'm now experiencing it when running Modelsim Altera on CentOS 7 (also on CentOS 6.5).

It would be nice if there was a workaround for this issue in vUnit. vUnit would have to detect that the vmap bug exists, and then modify 'modelsim.ini' without using the 'vmap' command.

Regards,

Daniel Blomkvist

kraigher, 2015-10-19:
Maybe it modifies some other modelsim.ini? We use `vmap` with the `-modelsimini` flag, maybe that does not work on your system? We have had some problems with modelsim.ini bugs and inconsistencies between versions.

I have made a possible fix in the [issue_80](https://github.com/LarsAsplund/vunit/tree/issue_80) branch were I change the current working directory before calling `vmap` rather than relying on the `-modelsimini` flag. Could you try it?

danielblomkvistbitvis, 2015-10-20:
I don't know exactly what the problem is, but it occurs in our own scripts even though we do not use the `-modelsimini` flag. I will try the issue_80 branch later today and get back to you.

danielblomkvistbitvis, 2015-10-20:
I've tried the new branch. Unfortunately it did not help.

I tried to find out if any other modelsim.ini had been edited, but there is only one other modelsim.ini in our case. It resides in ./opt/altera/14.0/modelsim_ase/modelsim.ini, and it had not been edited. 

As I mentioned this happens when we call vmap with our own scripts as well, so it's not a problem with vUnit itself. But it would be nice if we didn't have to think about it!

kraigher, 2015-10-20:
Is it bug present both if you run `vmap.exe` as a stand alone binary as well as run `vmap` as part of a Modelsim .do TCL script?

kraigher, 2015-10-20:
I have removed the use of vmap completely on the issue_80 branch. Instead the modelsim.ini file is manipulated directly from Python. Could you give it a try?

danielblomkvistbitvis, 2015-10-21:
The issue_80 branch now works for me. I can confirm that the bug is present even when I run vmap as a stand alone binary.

kraigher, 2015-10-21:
I have integrated the issue_80 branch into master and made a new release.

closed 2015-10-21 by kraigher