VUnit/vunit #81: vUnit support for encrypted VHDL source files (.vhdp)

danielblomkvistbitvis, 2015-10-19:
Hi,

I'm attempting to compile a project that includes some encrypted source files with file ending .vhdp.

I'm getting the following error:

RuntimeError: Unknown file ending '.vhdp'

I would like to have the possibility of using encrypted source files with vUnit. I realize that whether or not the project will run successfully depends on what parts that are encrypted, but it would be nice with file support.

Regards,

Daniel Blomkvist

kraigher, 2015-10-19:
Dependency scanning will never work with encrypted files since VUnit cannot decrypt them.
We would need to add a mechanism to add files without dependency scanning where the dependency is specified manually by the user. This is something we could easily do in the near future.

It still possible to use VUnit in a project containing encrypted files and I have done so in at least two projects. Typically encrypted source files are part of some third-party library or similar which can be compiled externally and added to VUnit using the `add_external_library` method. I typically pre-compile unisim and Vivado IP:s and add them as external libraries to VUnit.

danielblomkvistbitvis, 2015-10-20:
Alright. Thanks for the help. The `add_external_library` method is probably sufficient to solve my problem for now.

danielblomkvistbitvis, 2015-11-06:
Closing this issue since the work-around was OK.

closed 2015-11-06 by danielblomkvistbitvis