VUnit/vunit #83: Fix version in setup.py

go2sh, 2015-10-21:
Hello there,

I just tried to package vunit for debian for myself, but setup.py file has some errors. The version does not reflect the current tag and the version needs to start with a number e.g. "0.39.0".
And maybe you can upload it into the index. :)

Regards,
Chris

kraigher, 2015-10-21:
I would like to fix the @TODO in the setup.py with the data folder ending up outside the package in the python library root before adding it to the PIP index.

kraigher, 2015-10-22:
@LarsAsplund I have moved the `vhdl` and `verilog` folders into `vunit/hdl` to be compatible with the setuptools package concept where data files cannot be outside of the package. Check out the [move_data_files](https://github.com/LarsAsplund/vunit/tree/move_data_files) branch and see if you have any comments.

kraigher, 2015-10-22:
@LarsAsplund I renamed the folder `vunit/data` instead of `vunit/hdl` to make it more future proof for whatever data we want to ship with the python package.

kraigher, 2015-10-22:
@LarsAsplund have you made an account at PyPI such that you can be the "Owner" of VUnit there? I guess I also need an account there and you can set me as "Maintainer" such that I also can make new releases there.

go2sh, 2015-10-22:
@kraigher  I tested your branch with some of my projects. Works well.

LarsAsplund, 2015-10-23:
@kraigher My only comment is that we should use descriptive names for files and directories so I suggest we use `vunit/hdl` or remove the `hdl` directory level altogether. If we add some other files in the future they should be placed in equally descriptive directories.

I have an account on PyPI but I haven't registered anything yet. `vunit_hdl` is still free.

kraigher, 2015-10-25:
@LarsAsplund could you register vunit_hdl?

LarsAsplund, 2015-10-25:
@kraigher I've registered it now and set you as a maintainer as well. I few things caught my attention but I haven't figured out the all the details:
- I might be possible to redirect pip installs to the release on Github. Seems natural to have all traffic to one place but maybe there are some drawbacks?
- Currently a number of files are ignored when you register. Typically .md files and run.py files so the acceptance tests won't work after pip install
- http://peterdowns.com/posts/first-time-with-pypi.html indicates that having a name vunit_hdl that differs from the module name vunit may be a problem. He also shows how you make PyPI aware of README.md. I'm not sure about the purpose.

kraigher, 2015-10-26:
@LarsAsplund 
* I ignored the files to reduce the disk size of the Python package by not having pdf:s and png:s in there. Maybe it is better to not care about disk size and just add all files.
* Having two packages with different names on PyPI but with the same actual Python package name cause name conflicts when both packages are installed. For instance the users of the current `vunit` package could probably not install our `vunit_hdl` and have both working at the same time.

LarsAsplund, 2015-10-27:
@kraigher With the current size of the distribution I think the risk of someone missing documentation is higher than the risk of someone being annoyed by a few extra MBs so we should probably just keep everything. We should also have a way of verifying this type of distribution such that we don't forget to include some file. Not necessary to fix right now though. Another thing is that the OSVVM license requires documentation of the modifications we've made to make it GHDL compatible. This is only documented in the README file so I will see if I can figure out what the referenced tutorial meant by making PyPI aware of that file.

kraigher, 2015-10-28:
@LarsAsplund I am fine with adding everything and disregard size. We should however not think anyone will read the documentation in the form of .md and .pdf files when it is part of the Python package. The Python package is unpacked into a global Python install folder which no average user will look at to find documentation. Users installing the package via PyPI would still have to read their documentation here on GitHub.

LarsAsplund, 2015-10-28:
@kraigher Agree, it's not very likely. Hopefully, we can have the README shown on PyPI or we can simply add a link to the description field. Once OSVVM supports GHDL natively we can also remove the special notes on a the modified OSVVM distribution (assuming we include their PDFs).

LarsAsplund, 2015-10-28:
@go2sh  @kraigher  I changed our README.md to README.rst using reStructuredText markup which both PyPI and Github can render. I've uploaded it to [testpypi](https://testpypi.python.org/pypi/vunit_hdl/0.42.0). Could you give it a try and install it from there (ignore the version number, it will be 0.43.0 when I upload it to the official PyPI)?

``` console
pip install -i https://testpypi.python.org/pypi vunit_hdl
```

If it works I will go ahead and upload it officially

kraigher, 2015-10-29:
It works.

go2sh, 2015-10-29:
Works for me.

LarsAsplund, 2015-10-29:
There's now a v0.43.0 on PyPI.

commit 86a1e99a (2015-10-21): fix #83

* set the version number to 0.39.0
* removed the 'v' prefix, see PEP0440
closed 2015-10-29 by LarsAsplund