VUnit/vunit #85: generic packages

vermaete, 2015-10-29:
Could it be the VHDL parser can not handle the new/2008 generic packages?

kraigher, 2015-10-30:
The VUnit parser is lazy and only parses what it needs. Generic packages is not something we try to recognize specifically and they should be considered as regular packages. The generic part of them should be ignored. Are you seeing some crash or incorrect dependency or what lead up to this question?

pStefanNilsson, 2015-11-19:
The VUnit parser seams to crash if you declare a package as a generic to an entity. The same is probably true for type and function generics too. 

<pre>
entity generic_comp is
  generic (package generic_comp_pkg is new work.generic_pkg generic map (<>));
end generic_comp;
</pre>

kraigher, 2015-11-19:
Thank you pStefanNilsson for you problem description. It would be helpful if you also could provide explicit examples for other cases that you know do not work as well.

kraigher, 2015-11-22:
@pStefanNilsson, @vermaete and others... I have pushed a commit which causes package generics to be ignored rather than produce a crash. Could you verify that this solves your problems? If it does I will create a new VUnit release including the fix.

pStefanNilsson, 2015-11-23:
Thanks kraigher. I tried your fix and it works for me.
Sorry for a late answer on your previous question but below is a few more examples of new odd generics that I think the Vunit parser can't handle.

```vhdl
-- Example from:
-- VHDL-2008: Just the New Stuff
-- Peter J. Ashenden,Jim Lewis

entity generic_counter is
  generic ( type count_type;
            constant reset_value : count_type;
            function increment ( x : count_type ) return count_type );
  port ( clk, reset : in bit;
         data : out count_type );
end entity generic_counter;

begin
  count : process (clk) is
    begin
      if rising_edge(clk) then
        if reset = '1' then
          data <= reset_value;
        else
          data <= increment(data);
        end if;
      end if;
    end process count;
end architecture rtl;
```

kraigher, 2015-11-24:
@pStefanNilsson I have also added ignoring of type and function/procedure generics. Let me know if this is enough to close this issue.

pStefanNilsson, 2015-11-24:
Works fine!

 

Thanks,

Stefan

closed 2015-11-24 by kraigher