VUnit/vunit #86: Is is possible to set generics of type "TIME"?

gunnarstjernberg, 2015-11-18:
If I use a generic of type TIME in my testbench, it seems I cannot alter the value of said generic from the python script.

Example:

Defintiion at the TB top level design
g_clk_axi                         : TIME   := 4 ns; 

The python script statement:
tc_debug_0.set_generic("g_clk_axi", "4ns")

vsim message:
# ** Warning: (vsim-3351) Invalid value ""4ns"" for generic "g_clk_axi".  Not using this value.

The normal vsim use would be -gg_clk_axi=4ns but as vuint parses the above to -gg_clk_axi="4ns" it does not work.

Is tere any other way of controlling the TIME generic?

kraigher, 2015-11-18:
I can see if I can make it work. In the meantime you could make it a string generic and use `time'value` to convert to a time value in VHDL.

gunnarstjernberg, 2015-11-18:
Sounds good. I can us this work-around for the time being.  Thanks!

kraigher, 2015-11-21:
You should be able to set generics of type time now. Expect if you use GHDL.

closed 2015-11-21 by kraigher