VUnit/vunit #87: Using array_pkg with std_logic_vectors and unsigned

BjornOlsson, 2015-11-19:
At this point, when using vunit's array_pkg, it only supports integers as values. I think it would be a good thing to expand this to support other datatypes as well, such as std_logic, std_logic_vector, unsigned etc. 
It would also be great if the array_pkg could support records of these data types as values.

kraigher, 2015-11-19:
Expanding to std_logic_vector/signed/unsigned of more than 32 bits is possible and I will fix that. Arbitrary records would only be possible using VHDL-2008 generic packages, I will investigate.

LarsAsplund, 2016-06-13:
GHDL is currently lacking support for generic packages so such a feature can't be verified in Travis. Supported by ModelSim, Riviera-PRO and Active-HDL though.

LarsAsplund, 2016-06-15:
@BjornOlsson @kraigher Made some prototyping and the support provided  by ModelSim, Riviera-PRO and Active-HDL seems solid enough to make this work. The input to a generic array package is essentially the type and the functions to convert between values of that type and the string to store on file. We can use the predefined encode/decode functions from the `com` package for this as well as the generators of these for custom types. A generic package would also need a slightly different API since parameters like `bit_width` doesn't apply to all types.

bradleyharden, 2019-02-17:
It looks like GHDL now supports subprogram generics (albeit without default values). This should now be possible to implement. If I'm interested, should I start working on it? I can make a PR to discuss the implementation.

Also, I'd like to propose some way of retrieving a slice from an array, but I'm not sure of the best approach. Maybe new, overloaded functions get_slice and set_slice for 1D, 2D and 3D arrays?

LarsAsplund, 2019-02-18:
@bradleyharden Note that this issue concerns the older `array_pkg`. We should use the `interger_array_pkg`  as the starting point. For that reason I suggest that you start a new issue. Then I will close this one with a reference to the new

std-max, 2022-05-31:
@LarsAsplund
I think we can close this issue, as you mentionned it refers to array_pkg which support will be dropped in the next major release (I think I've read that somewhere?)

closed 2022-05-31 by LarsAsplund

umarcor, 2022-06-01:
@std-max, see https://github.com/VUnit/vunit/milestone/3 and https://github.com/VUnit/vunit/pull/600.