VUnit/vunit #91: GHDL needs –no-vital-checks option to compile Xilinx library

imd1, 2015-12-21:
I have code using the Xilinx Unisim library and must patch VUnit to include the –no-vital-checks option in all GHDL invocations. Is there a way to avoid patching VUnit and provide this option? I'm using GHDL 0.33, windows mcode.

kraigher, 2015-12-21:
Is is a compile time or run-time option or both? I have planned to add the ability to set tool specific compile flags on individual files and simulation related flags on test benches but not gotten around to it yet.

imd1, 2015-12-21:
As it's every invocation, I assume both.

kraigher, 2015-12-23:
I have made this possible using a combination of `set_sim_option` and `set_compile_option` in release *v0.49.0*.
See:
https://github.com/LarsAsplund/vunit/blob/master/user_guide.md#setting-custom-simulation-options
https://github.com/LarsAsplund/vunit/blob/master/user_guide.md#setting-custom-compilation-options

Just add this to the end of your `run.py` after adding all files:
```python
vu.set_compile_option("ghdl_flags", ["--no-vital-checks"])
vu.set_sim_option("ghdl_flags", ["--no-vital-checks"])
```

Merry Christmas!

closed 2015-12-23 by kraigher