VUnit/vunit #93: Missing files not reported

mark-newsam, 2016-01-07:
Hi,
While trying to set up my own design I made a mistake with path names, rather than getting a file not found error/warning I would get either:
   WARNING - otp_main.vhd: failed to find a primary design unit 'jtag_pkg' in library 'jtag'
Got this for getting the package file path wrong.

OR:
   # ** Warning: otp_main.vhd(130): (vopt-3473) Component instance "/tb_otp/i_otp_main/i_jtag_otp_tdr : jtag_tdr"
This is when an sub-module path is wrong.  It takes until the simulation starts to report a missing part which is due to filename issues.

Not being used to this software it took me a while to realise I had the path wrong.  Would it be possible to add a check to see if each source file actually exists and throw warnings/errors for each missing file.

Cheers,
   Mark

kraigher, 2016-01-07:
Good suggestion. The reason why no error is reported is due to the semantics of the `add_source_files` method. The argument is not a file name but a pattern which may contain wildcards. A single file name is also a pattern than can only match one thing. No files matching the pattern is currently not an error, even in the underlying Python standard library function. 

We could add an error when there are no files matching the pattern with an optional parameter to ignore the error. Some users might have some regular folder structure defined together with a general purpose python function to add them to VUnit where not all possible sub-folders exist or contain files for every module and then they do not want to have errors. As you say having the error by default is probably the best option.

kraigher, 2016-01-07:
Fixed in release v0.50.0

closed 2016-01-07 by kraigher