VUnit/vunit #96: VUnit fails to detect ModelSim

igmar, 2016-01-16:
Hi,

VUnit fails to detect modelsim, which is in my path : 

igmar@cad:~$ which vsim
/usr/local/bin/vsim

If I run it from the shell, it opens up the ModelSim GUI. This is the Altera Starter Edition in case that is relevant. I suspect it can't find the modelsim.ini file, but that is located in the install dir. It might have an issue finding it, since the vsim in /usr/local/bin is a symlink to a location outside of PATH.

We might want to improve the error message in this case.

kraigher, 2016-01-16:
The simulator install detection mechanism became slightly more complicated when we added support for Active-HDL and Riviera-Pro. Before we just checked for vsim on the PATH. But people have both installed and then we need to tell the difference so we also check for modelsim.ini to detect modelsim. I think the problem could be solved by following the symlink to the target file and check for modelsim.ini there.

kraigher, 2016-01-16:
I will make a fix for this. In the meantime a work-around would be to add the original modelsim install directory to the PATH before /usr/local/bin. For instance on my machine I have `/opt/altera/14.1/modelsim_ase/bin/` in the PATH.

igmar, 2016-01-16:
Fixing the path indeed fixed the issue.

kraigher, 2016-01-17:
@igmar Was the symbolic link in `/usr/local/bin` the result of an official install process or something you created yourself?

igmar, 2016-01-18:
I created that myself. I usually have a tools linked in /usr/local/bin, since I prefer a prompt over a click in some gui. Modelsim itself seems find with that, vunit then fails to discover the modelsim.ini file.

kraigher, 2016-01-18:
@igmar You can check the issue_96 branch. It should try to detect the modelsim toolchain path even in your case now. Do not forget to remove the PATH modification you made when testing if it works.

LarsAsplund, 2016-06-09:
@igmar Has this issue been solved?

LarsAsplund, 2016-10-05:
Closing this issue due to lack of information.

closed 2016-10-05 by LarsAsplund