VUnit/vunit #97: mixed mode simulations

swattor, 2016-01-20:
I am trying to run a mixed-mode (SV and VHDL) simulation on modelsim through vunit. 

I have certain VHDL packages that need to be compiled with the -mixedsvvh flag to allow systemverilog testbench to look at contents - but there doesn't appear to be an easy way of adding a "per file" compilation option.

kraigher, 2016-01-20:
We do have a concept for per file based compilation options. See `set_compile_option` in the user guide. We have only added hooks for GHDL though but I will also add one for Modelsim vcom to solve your needs.

So you are running SystemVerilog testbenches through VUnit? You must like to be on the bleeding edge. SV support is not something we even officially have yet and the hidden support we do have is not complete and we do not promise that it will be stable. We have the ambition to make VUnit work just as nice with SV though.
Do not get me wrong, I am happy to have someone wanting to use SV with VUnit. Maybe you can help us shape such support by beta testing it?

swattor, 2016-01-20:
@kraigher funny enough - have just made the same mod - having discovered the set_compile_option!

kraigher, 2016-01-20:
Added `modelsim_vcom_flags` and `modelsim_vlog_flags` and released `v0.51.0`

closed 2016-01-20 by kraigher

swattor, 2016-01-20:
That's great. I've noticed that the dependency checking has some issues as well - i.e. if a verilog file requires a VHDL package... I'm looking into it now, but is this something that you already have some ideas on?

kraigher, 2016-01-20:
There is no dependency checking between VHDL and Verilog at the moment. 
Only dependencies internally in the individual VHDL and Verilog domains.
The reason is that this has not been added yet is the assumption that people either use only VHDL or only Verilog. Maybe you can create a new issue where you suggest suitable rules for dependencies between VHDL and Verilog?  Sometime in the future I also want to add support for the user to manually declare dependencies between files.

swattor, 2016-01-20:
Sure, I can do that - no problem.

kraigher, 2016-01-24:
@swattor I have added a way to specify manual dependencies.