VUnit/vunit #100: Add --debug argument

cmarqu, 2016-01-29:
Hi,

I'm looking for a way to selectively print debug output for the simulator setup, like the command line used etc.

There is the `--verbose` argument but that does seems to be reserved to control the verbosity of the simulation run itself. So I'd propose a globally available `--debug` argument.

kraigher, 2016-01-29:
We already have `--log-level=debug`?

cmarqu, 2016-01-29:
Ah, indeed, I managed to miss that. Maybe because it doesn't have a help string - could you please add one?

kraigher, 2016-01-29:
I have already added a help text in the `sphinx`_doc branch were we are working on taking the VUnit documentation to the next level. It will come to master soon.

cmarqu, 2016-01-29:
Cool, thanks!

closed 2016-01-29 by cmarqu