VUnit/vunit #104: ghdl - compile_option is not consistently passed to ghdl

Xaerdna, 2016-02-01:
Added ui.set_compile_option("ghdl_flags", ["-frelaxed-rules"]) to the execution script, which produced the following output:

Compiling /root/workspace/test/vunit/vunit/vhdl/string_ops/src/string_ops.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/vhdl/src/lib/std/textio.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/run/src/stop_api.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/run/src/stop_body_2008.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/vhdl/src/lang/lang.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/path/src/path.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/logging/src/log_types.vhd into vunit_lib ...
/root/workspace/test/vunit/vunit/vhdl/logging/src/log_types.vhd:64:14:warning: package "log_types_pkg" does not require a body
Compiling /root/workspace/test/vunit/vunit/vhdl/logging/src/log_formatting.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/logging/src/log_special_types200x.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/logging/src/log_base_api.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/logging/src/log_api.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/check/src/check_types.vhd into vunit_lib ...
/root/workspace/test/vunit/vunit/vhdl/check/src/check_types.vhd:36:14:warning: package "check_types_pkg" does not require a body
Compiling /root/workspace/test/vunit/vunit/vhdl/logging/src/log.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/check/src/check_special_types200x.vhd into vunit_lib ...
Compiling /root/workspace/test/uvvm_util/src/types_pkg.vhd into uvvm_util ...
/root/workspace/test/uvvm_util/src/types_pkg.vhd:120:14:warning: package "types_pkg" does not require a body
Compiling /root/workspace/test/uvvm_util/src/adaptations_pkg.vhd into uvvm_util ...
/root/workspace/test/uvvm_util/src/adaptations_pkg.vhd:206:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/adaptations_pkg.vhd:267:14:warning: package "adaptations_pkg" does not require a body
Compiling /root/workspace/test/uvvm_util/src/string_methods_pkg.vhd into uvvm_util ...
/root/workspace/test/uvvm_util/src/adaptations_pkg.vhd:206:19:warning: type of a shared variable must be a protected type
Compiling /root/workspace/test/uvvm_util/src/license_pkg.vhd into uvvm_util ...
/root/workspace/test/uvvm_util/src/adaptations_pkg.vhd:206:19:warning: type of a shared variable must be a protected type
Compiling /root/workspace/test/uvvm_util/src/hierarchy_linked_list_pkg.vhd into uvvm_util ...
/root/workspace/test/uvvm_util/src/adaptations_pkg.vhd:206:19:warning: type of a shared variable must be a protected type
Compiling /root/workspace/test/uvvm_util/src/protected_types_pkg.vhd into uvvm_util ...
/root/workspace/test/uvvm_util/src/adaptations_pkg.vhd:206:19:warning: type of a shared variable must be a protected type
Compiling /root/workspace/test/uvvm_util/src/alert_hierarchy_pkg.vhd into uvvm_util ...
/root/workspace/test/uvvm_util/src/adaptations_pkg.vhd:206:19:warning: type of a shared variable must be a protected type
Compiling /root/workspace/test/uvvm_util/src/methods_pkg.vhd into uvvm_util ...
/root/workspace/test/uvvm_util/src/adaptations_pkg.vhd:206:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:39:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:40:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:41:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:42:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:43:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:44:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:45:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:46:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:47:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:48:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:49:19:warning: type of a shared variable must be a protected type
Compiling /root/workspace/test/uvvm_util/src/bfm_common_pkg.vhd into uvvm_util ...
/root/workspace/test/uvvm_util/src/adaptations_pkg.vhd:206:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:39:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:40:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:41:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:42:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:43:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:44:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:45:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:46:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:47:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:48:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:49:19:warning: type of a shared variable must be a protected type
Compiling /root/workspace/test/uvvm_util/src/uvvm_util_context.vhd into uvvm_util ...
/root/workspace/test/uvvm_util/src/adaptations_pkg.vhd:206:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:39:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:40:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:41:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:42:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:43:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:44:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:45:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:46:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:47:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:48:19:warning: type of a shared variable must be a protected type
/root/workspace/test/uvvm_util/src/methods_pkg.vhd:49:19:warning: type of a shared variable must be a protected type
Compiling /root/workspace/test/vunit/vunit/vhdl/check/src/check_base_api.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/check/src/check_api.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/check/src/check.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/check/src/check_base.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/logging/src/log_base.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/dictionary/src/dictionary.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/run/src/run_types.vhd into vunit_lib ...
Compiling /root/workspace/test/vunit/vunit/vhdl/run/src/run_special_types200x.vhd into vunit_lib ...
[... cut ...]
Compiling /root/workspace/test/vunit/vunit/vhdl/vunit_run_context.vhd into vunit_lib ...
Compiling /root/workspace/test/uvvm_util/tb/methods_tb.vhd into uvvm_util ...
/root/workspace/test/uvvm_util/src/adaptations_pkg.vhd:206:19: type of a shared variable must be a protected type

"type of a shared variable must be a protected type" is considered a warning, until it suddenly produces an error.

Adding '-frelaxed-rules' directly to the command-list in compile_vhdl_file() in ghdl_interface.py caused the program to execute correctly.

kraigher, 2016-02-01:
Seems like you added more files after setting the option? The option applies to everything you have already added but not future additions.

kraigher, 2016-02-01:
looks like uvvm stuff was added after the option was set.

Xaerdna, 2016-02-02:
Ah, ok. Missed that part. Thanks!

closed 2016-02-02 by Xaerdna