VUnit/vunit #106: user_guide example fails on OS X with binary ghdl

twam, 2016-02-01:
I installed the latest OS X binary from ghdl (ghdl-0.31-gcc-darwin13.mpkg) and tried to get the user_guide example running. 

It fails with
```
Compiling /Users/twam/Temp/vunit_virtualenv/lib/python3.5/site-packages/vunit/vhdl/vhdl/src/lang/lang.vhd into vunit_lib ...
/opt/ghdl/gcc-4.8.2/libexec/gcc/x86_64-apple-darwin13.0.0/4.8.2/ghdl1: cannot find "std" library
ghdl: compilation error
```

All the examples from http://ghdl.readthedocs.org/en/latest/Starting_with_GHDL.html work with GHDL, so I reckon that the problem is on vunit's side.

kraigher, 2016-02-01:
GHDL 0.31 is to old you need >= 0.33

twam, 2016-02-01:
That was quick and helpful. Installed the ghdl-0.33-llvm-darwin13.pkg (wasn't advertised at the front page :() and now it works!

closed 2016-02-01 by twam