VUnit/vunit #108: Vivado and VUnit example with ghdl

twam, 2016-02-03:
The "Vivado and VUnit example" does not work for me:

```
Compiling unisim library into .../vunit/examples/vhdl/vivado/unisim ...
Creating library unisim in unisim
Traceback (most recent call last):
  File "run.py", line 22, in <module>
    add_vivado_ip(ui, root, project_file=join(root, "myproject", "myproject.xpr"))
  File ".../vunit/examples/vhdl/vivado/vivado_util.py", line 123, in add_vivado_ip
    compile_unisim(unisim_library_path)
  File ".../vunit/examples/vhdl/vivado/vivado_util.py", line 46, in compile_unisim
    create_library("unisim", library_path, library_path)
  File ".../vunit/examples/vhdl/vivado/vivado_util.py", line 28, in create_library
    check_call(['vlib', '-unix', library_path])
  File "/usr/lib/python3.4/subprocess.py", line 556, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python3.4/subprocess.py", line 537, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.4/subprocess.py", line 859, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.4/subprocess.py", line 1457, in _execute_child
    raise child_exception_type(errno_num, err_msg)
FileNotFoundError: [Errno 2] No such file or directory: 'vlib'
```

I guess the problem is that create_library function trys to call some Modelsim specific stuff, however I like to use it with ghdl.

kraigher, 2016-02-03:
The basic idea is to have a script which compiles the Vivado standard libraries like `unisim` as well as the project specific IP libraries *outside* of VUnit and then add those as external libraries to VUnit. This is how I have worked with Vivado in two different projects and taught some other people to do as well. It allows you to just check in the `.xci` files or the `.tcl` files which generates the IP:s into version control and then ask Vivado to generate the simulation models and their compile order. This external compile is done outside of VUnit and thus it uses commands hardcoded to a certain simulator which is modelsim in this case. Typically Vivado IP:s contain encrypted files which VUnit cannot scan for dependencies and GHDL cannot compile. 

It has been a while since I created this example, maybe I can update it to support more simulators.

twam, 2016-02-03:
Thanks for the quick reply. I would be awesome if you could improve this example. As a first step I would be beneficent (for me) to use unisim stuff, like block ram, dcm or dsp slices. As I'm using this as a hobbyist ghdl is basically the only affordable solution :) and as far as I know many of the IPs also cost a fortune to license.

kraigher, 2016-02-03:
Well just checked. Even the simple fifo IP I have included in the Vivado example uses encrypted VHDL which GHDL will never be able to compile for legal reasons.

kraigher, 2016-02-03:
Unisim should be able to be compiled with GHDL though.

kraigher, 2016-02-03:
@twam Compiling unisim with GHDL is mostly about finding the correct compile flags for GHDL to get it to ignore the many LRM violations in the Xilinx code. It is not something I can help much with. Try looking here: https://github.com/tgingold/ghdl

twam, 2016-02-03:
Would it then be possible to use Xilinx's ISim?

kraigher, 2016-02-03:
VUnit does not support Xilinx isim or Vivado xsim at the moment.

twam, 2016-02-03:
Alright! Then I'll hope for the future :) Thanks for your quick and detailed reply!

ferdepe, 2016-05-18:
Hi! I executed the same example as @twam and I have the same errors... I haven't got modelsim neither. Sorry, but why is it no possible with GHDL??

LarsAsplund, 2016-06-09:
I think this question has been answered and can be closed.

closed 2016-06-09 by LarsAsplund