VUnit/vunit #109: Support Verilog-AMS files

cmarqu, 2016-02-04:
Verilog-AMS files are often used for modelling analog behaviour because of their `wreal` port type.
Please add simple support for that.
Initially, this requires accepting `*.vams` files, and there are ``include "constants.vams"` etc. statements that may not resolve to real files in the filesystem (but https://github.com/VUnit/vunit/blob/master/vunit/parsing/verilog/preprocess.py#L47 fails without a good error message when it's not finding them).

LarsAsplund, 2016-06-15:
@cmarqu @kraigher What's the status on this one? Should it be closed or are there planned extensions to this?

cmarqu, 2016-06-15:
They are supported now (thanks!), so I'm closing.

closed 2016-06-15 by cmarqu