VUnit/vunit #113: Verilog parser problem

cmarqu, 2016-02-05:
I have a file (most likely from an IP block) that is producing a parser error:
```
Traceback (most recent call last):
  File "/home/marquardt/tools/vunit/vunit/project.py", line 626, in parse
    design_file = parser.parse(code, self.name, include_dirs, self._content_hash)
  File "/home/marquardt/tools/vunit/vunit/parsing/verilog/parser.py", line 28, in parse
    return VerilogDesignFile.parse(code, file_name, include_paths)
  File "/home/marquardt/tools/vunit/vunit/parsing/verilog/parser.py", line 56, in parse
    included_files=included_files)
  File "/home/marquardt/tools/vunit/vunit/parsing/verilog/preprocess.py", line 39, in preprocess
    file_name = stream.pop().value
AttributeError: 'NoneType' object has no attribute 'value'
Traceback (most recent call last):
  File "/home/marquardt/tools/vunit/vunit/project.py", line 626, in parse
    design_file = parser.parse(code, self.name, include_dirs, self._content_hash)
  File "/home/marquardt/tools/vunit/vunit/parsing/verilog/parser.py", line 28, in parse
    return VerilogDesignFile.parse(code, file_name, include_paths)
  File "/home/marquardt/tools/vunit/vunit/parsing/verilog/parser.py", line 56, in parse
    included_files=included_files)
  File "/home/marquardt/tools/vunit/vunit/parsing/verilog/preprocess.py", line 49, in preprocess
    with open(full_name, "r") as fptr:
IOError: [Errno 21] Is a directory: u'/local/proj1947/work/marquardt/designdata/rev_work/resources/ftp/Front-end/behavioral_model/'
```

cmarqu, 2016-02-05:
I traced it down to a file with a module header like this:
```
`celldefine
module `IP_NAME (
        input                                                       IP_EN,
        `ifdef N_ROWS_IS_1
        `else
            input           `ADDR_bus_decl                          ADDR,
        `endif
        input                                                       READ,
 ```

kraigher, 2016-02-05:
The first error is a file with an `include` that is not followed by a string. The second error is an `include` that specifies a directory and not a file.

cmarqu, 2016-02-05:
Ok, the first error indeed is written like `include `DEFINEFILE - maybe DEFINEFILE is not set correctly. Let me investigate that.

kraigher, 2016-02-05:
I does not handle ```include`` where the argument is not a string. We need to add the possibility to expand a define there as well.

kraigher, 2016-02-06:
The parser should now provide significantly better error messages as well as be more robust.

closed 2016-02-06 by kraigher